Arksine/moonraker #845: unable get printer info from moonraker

marcokitt2000, 2024-04-29:
Hello,

I have a sovol sv07 on my old klipper rpi screen hw 1.0 now allso with new version hw 1.2 i get every time i power on the system or reboot get this error.....
unable get printer info from moonraker but mainsail and fluidd works fine printer it prints only klipper screen on the screen comes retry button than it works fine it starts, or on mainsail on the upper right corner on/off button than service klipper screen restart icoon or stop than start(play buttom) it works fine. this was in begin with the usb deb update 1.0.17 or a full recovery from them 1.0.17 intermittent some times it past sometimes i must press retry. only the klipper screen for the rest works fine.
I think iget printer info from moonraker but it must wait little bit longer how can i add or more time before retry auto startup maybe it must wait what longer.

I hope in config files settings can do the trick my software on rpi klipper is OS: Armbian 22.05.0-trunk Buster maybe thats a issue.

Gr. Marco i like klipper,moonraker love it by the 1st saw and use.

marcokitt2000, 2024-04-29:
I did a check i used from Christian Vick kpa 3.0b91 when i disabled with ; on printer.cfg file its starts normal still with the txt klipper logo unable to get printer info from moonraker but than the screen goes on. i think the time it must wait before a auto replay(maybe 1x instead off 3x times) can we adjust it where and how?

Arksine, 2024-04-30:
Hi Marco, my understanding is that the Sovol SV7 runs a fork of Moonraker.  I would suggest contacting them for support.

At a high level, if the frontend is requesting `/printer/info` before Moonraker has had a chance to connect to Klipper then that is an issue that would need to be resolved in the frontend.  It needs to handle the error and and re-request if necessary.

marcokitt2000, 2024-04-30:
Thank you for yours response, without Christian Vicks KPA it works fine startups fine.
https://www.printables.com/model/585691-klipper-printer-additions-30b8-more-than-boring-st
But i updated to latest versions.....
[image]
check mine screenshot normal sv07 firmware release 1.0.17 or the full image is klipper v11 i use now v12.0.173 i now its now 177.
i send my klippy log and moonraker log files to.
[moonraker.log](https://github.com/Arksine/moonraker/files/15169770/moonraker.log)
[klippy.log](https://github.com/Arksine/moonraker/files/15169772/klippy.log)

Maybe someone now a trick.

Thnxxxxxx but it prints.

marcokitt2000, 2024-05-04:
I have update to bookwurm here on github and works fine nice. thanks for evrything keep the good working klipper moonraker to work.

Gr. Marco

Arksine, 2024-05-08:
As this appears to have been resolved I'm going to close this.

closed 2024-05-08 by Arksine

ImhoFpV, 2024-09-20:
> I have update to bookwurm here on github and works fine nice. thanks for evrything keep the good working klipper moonraker to work.
> 
> Gr. Marco

How you update to bookwurm??

marcokitt2000, 2024-09-21:
Yes, sovol sv07 (i think it will work fir sv07 plus is the same klipad but bjgger bed printer.cfg file).
I will post a link..... if you have sovol sv07 and has the mks v2 us emmc dongle its easy to do. On the link there is a link to img file from wo make it posible for us.

Later mail will be come...

Gr. Marco

marcokitt2000, 2024-09-21:
https://github.com/vasyl83/SV07update

The copy trick over the 2 usb was for me not working i have copy img direct to emmc.
Than for me works fine.
If you have sovol sv07 i can upload img file there is all so kpa 3.0b91 from Christian Vick.

ImhoFpV, 2024-09-21:
> https://github.com/vasyl83/SV07update
> 
> El truco de copiar a través de los 2 USB no me funcionó. Copié la imagen directamente a EMMC. Entonces, para mí, funciona bien. Si tienes SOVOL SV07, puedo cargar el archivo img. Todo está en KPA 3.0B91 de Christian Vick.

Hello 

Awesome friend, thanks for this
I followed the steps of the 2 usb trick, because flashing directly on the emmc didn't work for me.
It is a bit more tedious, but it seems to work.

Thanks

ImhoFpV, 2024-09-21:
> Yes, sovol sv07 (i think it will work fir sv07 plus is the same klipad but bjgger bed printer.cfg file). I will post a link..... if you have sovol sv07 and has the mks v2 us emmc dongle its easy to [...]

Hello

Awesome friend, thanks for this
I followed the steps of the 2 usb trick, because flashing directly on the emmc didn't work for me.
It is a bit more tedious, but it seems to work.

Thanks

ImhoFpV, 2024-09-21:
> https://github.com/vasyl83/SV07update
> 
> The copy trick over the 2 usb was for me not working i have copy img direct to emmc. Than for me works fine. If you have sovol sv07 i can upload img file there is all so kpa 3.0b91 from Christian Vick.

I have the sv07 
It would be great if you could send me the img file.

Thanks

marcokitt2000, 2024-09-21:
please sent a email adres....

ImhoFpV, 2024-09-21:
> please sent a email adres....

gchinea76@gmail.com

Thanks

marcokitt2000, 2024-09-21:
https://netztorte.de/3d/doku.php?id=image
 he build a img file on it. you can download there but kpa is not in it.  i will upload my imahe with kpa in it.

marcokitt2000, 2024-09-21:
wait i must split the file in more peaces its bigger than 2gb i can sent only 2 gbmax for free
if you have 16gb instead of 8gb emmc than this is all ready work for it. other ways you must resize it to 8gb.
the other link from totzren is withou kpa and for 8gb emmc you can download it.
its all so ok.

ImhoFpV, 2024-09-21:
> espera, debo dividir el archivo en más partes, es más grande que 2 GB, solo puedo enviar 2 GB como máximo de forma gratuita si tienes 16 GB en lugar de 8 GB EMMC, entonces todo esto está listo para funcionar. De lo contrario, debes cambiar su tamaño a 8 GB. El otro enlace de totzren es sin kpa y para 8 GB EMMC puedes descargarlo. Todo está bien.

I have a 32gb emmc

marcokitt2000, 2024-09-21:
i am busy uploaded to wetransfer when done i sent link

ImhoFpV, 2024-09-21:
> i am busy uploaded to wetransfer when done i sent link

Don't worry, I'm in no hurry
I appreciate it enormously

marcokitt2000, 2024-09-21:
10% 1 min.