Arksine/moonraker #847: Gcode-shell-command Error: Failed to restart KlipperScreen.service: Interactive authentication required.

fabiandj64, 2024-05-01:
### What happened

When i try to restart the KlipperScreen.service with a script which runs when the given macro is used. i get the following error: "Failed to restart KlipperScreen.service: Interactive authentication required. See system logs and 'systemctl status KlipperScreen.service' for details." 

it worked on my old system a few weeks back. but i had to set up a new raspberry because i broke the fw on the last one :/ since setting up everything the command will no longer work. i tried everything i could find. added polkit rules, added sudoers, added every user to the NOPASSWD rule it changed nothing...

i dont know which files are needed for this problem but i can upload the requested files then

### Client

Fluidd, KlipperScreen

### Browser

Firefox

### How to reproduce

install gcode shell command 
make script including "systemctl restart KlipperScreen.service"
try to run script

fabiandj64, 2024-05-01:
when i try to run the script on the pi i get the following in the console  (hope it helps)

==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ====
Authentication is required to restart 'KlipperScreen.service'.
Authenticating as: ,,, (pi)
Password:

Arksine, 2024-05-08:
This doesn't appear related to Moonraker.  GCode Shell Command is a 3rd party modification to Klipper and not supported in the official repos.  I would suggest seeking assistance on one of the various discords, as someone there may be able/willing to assist with unofficial code.

closed 2025-01-20 by Arksine