Arksine/moonraker #852: Fluidd Moonraker connection to elegoo Neptune 4+ on MAC failed (Force Refresh)

ddiers, 2024-05-07:
### What happened

I have an elegy Neptune 4+ 3D-printer with Wifi connection. When I enter the IP on MAC via Safari or Chrome the below picture is shown and I don't get any connection. Also when I load the site new or click on Force Refresh nothing happens.
When I do exactly the same on my Android phone with Chrome it works, I directly access the printer webgui.

Is there any bug or what do I have to do to get also access from my MAC?

[image]

Many thx
Daniel

### Client

Fluidd

### Browser

Chrome, Safari

### How to reproduce

Type in IP in Safari or Chrome Browser on MAC

Daddiola1966, 2024-05-09:
I got the same issue. at about 4:30am eastern time, it suddenly stopped connecting.  Doesn't work in any browser but my primary is chrome

Arksine, 2024-05-09:
These printers tend to ship with forks of Moonraker, Klipper, etc.   If you know you are running an official version of Moonraker from this repo I'm willing to take a look at `moonraker.log` if you attach it.  Otherwise I would suggest reaching out to the manufacturer for support.

ddiers, 2024-05-22:
Thx, Here ist the log. Any Idea? @Arksine 
[moonraker.log.txt](https://github.com/Arksine/moonraker/files/15407964/moonraker.log.txt)

Arksine, 2024-05-27:
The log indicates that the version of Moonraker running on the machine is a fork.  Unfortunately I am unable to provide any guidance as I don't know what changes they could have made to Moonraker. 

That said, based on the description I'm skeptical that issue here is on the host.  It sounds like a browser/OS compatibility issue.

TheDuke22, 2024-06-14:
I think the issue is browser related (maybe security related?), I have same issue with another printer (creality K1), tried with MS edge and work without any issue

closed 2025-01-20 by Arksine