Arksine/moonraker #860: Power device: add option to run custom gcode on on/off action

PhilippMolitor, 2024-05-21:
### Is your feature request related to a problem? Please describe

When I power my device up or down (in my case, the 24V supply for motors and heaters), I want to have an option to run custom gcode before shutting off, and after turning on.

### Describe the solution you'd like

Add the following options to any power device:

```ini
# macro block to execute after turning on the power device
on_gcode_gcode:
  MY_CUSTOM_OFF_MACRO
  M83
# run this macro 3 seconds after turning on
on_gcode_delay: 3
# macro block to execute after turning off the power device
on_gcode_gcode:
  MY_CUSTOM_ON_MACRO
  SOME_OTHER_MACRO
# run this macro 3 seconds before turning off
on_gcode_delay: 3
```

### Describe alternatives you've considered

Adding a second power device for klipper does not seem reasonable, as I have to switch on two different devices then in the correct order.

Chomikolew, 2025-01-27:
That's a great idea; it would solve my problem with turning off addressable LEDs. I hope I'll see its implementation soon.

LazaroFilm, 2025-04-07:
Turning off leds before shutting down is that lead me here too.