Arksine/moonraker #863: [database.py:insert_item()] - Error inserting key 'instance_id' in namespace 'moonraker'

dbarwikowski, 2024-05-27:
### What happened

After moonraker update I receive this error. 
I host it on `Lenovo Ideacentre Stick 300 PC Mini Desktop`. I installed there `Ubuntu 18.04.6 LTS`.
Everything was running OK before update.

### Client

Mainsail

### Browser

Chrome

### How to reproduce

I was ~100 version behind. Updated using mainsail.

### Additional information

-------------------- Log Start | Mon May 27 18:14:19 2024 --------------------
platform: Linux-5.4.0-150-generic-x86_64-with-glibc2.26
data_path: /home/dbarwikowski/printer_data
is_default_data_path: False
config_file: /home/dbarwikowski/printer_data/config/moonraker.conf
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
instance_uuid: a4b6e48f6d594dedabc71ee4eb70c645
unix_socket_path: /home/dbarwikowski/printer_data/comms/moonraker.sock
software_version: v0.8.0-374-g5332eab
git_branch: master
git_remote: origin
git_repo_url: https://github.com/Arksine/moonraker.git
modified_files: []
unofficial_components: []
log_file: /home/dbarwikowski/printer_data/logs/moonraker.log
python_version: 3.7.5 (default, Dec  9 2021, 17:04:37)  [GCC 8.4.0]
launch_args: /home/dbarwikowski/moonraker-env/bin/python /home/dbarwikowski/moonraker/moonraker/moonraker.py -d /home/dbarwikowski/printer_data
msgspec_enabled: False
uvloop_enabled: False
2024-05-27 20:14:19,942 [confighelper.py:read_file()] - Configuration File '/home/dbarwikowski/printer_data/config/moonraker.conf' parsed, total size: 593 B
2024-05-27 20:14:19,943 [server.py:add_log_rollover_item()] -
#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
klippy_uds_address = /home/dbarwikowski/printer_data/comms/klippy.sock

[authorization]
trusted_clients =
        10.0.0.0/8
        127.0.0.0/8
        169.254.0.0/16
        172.16.0.0/12
        192.168.0.0/16
        FE80::/10
        ::1/128
        192.168.0.0/24
        192.168.1.0/24
cors_domains =
        *.lan
        *.local
        *://localhost
        *://localhost:*
        *://my.mainsail.xyz
        *://app.fluidd.xyz

[octoprint_compat]

[history]

[update_manager]
channel = dev
refresh_interval = 168

[update_manager mainsail]
type = web
channel = stable
repo = mainsail-crew/mainsail
path = ~/mainsail

#################################################################
All Configuration Files:
/home/dbarwikowski/printer_data/config/moonraker.conf
#################################################################
2024-05-27 20:14:20,053 [server.py:load_component()] - Component (secrets) loaded
2024-05-27 20:14:20,061 [server.py:load_component()] - Component (template) loaded
2024-05-27 20:14:20,070 [server.py:load_component()] - Component (klippy_connection) loaded
2024-05-27 20:14:20,228 [server.py:load_component()] - Component (application) loaded
2024-05-27 20:14:20,260 [server.py:load_component()] - Component (websockets) loaded
2024-05-27 20:14:20,311 [server.py:load_component()] - Component (dbus_manager) loaded
2024-05-27 20:14:20,322 [server.py:add_log_rollover_item()] - Loading Sqlite database provider. Sqlite Version: 3.22.0
2024-05-27 20:14:20,323 [database.py:insert_item()] - Error inserting key 'instance_id' in namespace 'moonraker'
2024-05-27 20:14:20,324 [server.py:add_log_rollover_item()] - Unsafe Shutdown Count: 0
2024-05-27 20:14:20,324 [database.py:insert_item()] - Error inserting key 'database_version' in namespace 'database'
2024-05-27 20:14:20,327 [server.py:load_component()] - Component (database) loaded
2024-05-27 20:14:20,349 [database.py:insert_item()] - Error inserting key 'protected_namespaces' in namespace 'database'
2024-05-27 20:14:20,353 [database.py:insert_item()] - Error inserting key 'file_manager.metadata_version' in namespace 'moonraker'
2024-05-27 20:14:20,355 [file_manager.py:__init__()] - Using File System Observer: inotify
2024-05-27 20:14:20,391 [server.py:load_component()] - Component (file_manager) loaded
2024-05-27 20:14:20,434 [database.py:register_table()] - Found registered table authorized_users
2024-05-27 20:14:20,437 [server.py:load_component()] - Unable to load component: (authorization)
Traceback (most recent call last):
  File "/home/dbarwikowski/moonraker/moonraker/server.py", line 297, in load_component
    component = load_func(config)
  File "/home/dbarwikowski/moonraker/moonraker/components/authorization.py", line 936, in load_component
    return Authorization(config)
  File "/home/dbarwikowski/moonraker/moonraker/components/authorization.py", line 150, in __init__
    self.user_table = database.register_table(UserSqlDefinition())
  File "/home/dbarwikowski/moonraker/moonraker/components/database.py", line 522, in register_table
    self.db_provider.register_table(table_def)
  File "/home/dbarwikowski/moonraker/moonraker/components/database.py", line 1295, in register_table
    table_def.name, table_def.prototype, table_def.version
  File "/home/dbarwikowski/moonraker/moonraker/components/database.py", line 870, in _save_registered_table
    (table_name, prototype, version)
sqlite3.OperationalError: near "ON": syntax error
2024-05-27 20:14:20,438 [server.py:main()] - Moonraker Error
Traceback (most recent call last):
  File "/home/dbarwikowski/moonraker/moonraker/server.py", line 662, in main
    server.load_components()
  File "/home/dbarwikowski/moonraker/moonraker/server.py", line 261, in load_components
    self.load_component(config, component)
  File "/home/dbarwikowski/moonraker/moonraker/server.py", line 297, in load_component
    component = load_func(config)
  File "/home/dbarwikowski/moonraker/moonraker/components/authorization.py", line 936, in load_component
    return Authorization(config)
  File "/home/dbarwikowski/moonraker/moonraker/components/authorization.py", line 150, in __init__
    self.user_table = database.register_table(UserSqlDefinition())
  File "/home/dbarwikowski/moonraker/moonraker/components/database.py", line 522, in register_table
    self.db_provider.register_table(table_def)
  File "/home/dbarwikowski/moonraker/moonraker/components/database.py", line 1295, in register_table
    table_def.name, table_def.prototype, table_def.version
  File "/home/dbarwikowski/moonraker/moonraker/components/database.py", line 870, in _save_registered_table
    (table_name, prototype, version)
sqlite3.OperationalError: near "ON": syntax error
2024-05-27 20:14:20,440 [server.py:main()] - Server Shutdown

dbarwikowski, 2024-05-27:
I've downgraded ~100 commits using KIAUH and moonraker works again

Arksine, 2024-05-27:
Unfortunately your OS is too far out of date for a recent change to Moonraker.   You can choose to stay on the current version or update your OS to something more recent.

dbarwikowski, 2024-05-27:
Thanks for reply. Would like to know which dependency is holding me back as upgrade is not an option. I guess sqlite?

dbarwikowski, 2024-05-27:
I've downgraded to  v0.8.0-360
[image]
Something in 361 breaks it for me.

Arksine, 2024-05-27:
Yes, your current issue is sqlite.  Version 3.24 or greater is required.  FWIW, Moonraker will drop support for Python 3.7 as well later this year.

eduncan911, 2024-10-11:
Some of us have long running micro processors running Python 3.8.  Dropping support would expel from our open devices in several labs.

1. Why not bundle sqlite into the install process?  KIAUH already does nginx.
2. Use an alternative datastore, something file based with no dependencies.  toml is my goto.

Ps, I was just hit with this today across 7 machines.  There's no way I have time to go re-setup each device from scratch on the custom builds.  They are on air-gapped networks, so no need for internet-based security.  

So simply "updating the OS" is not a friendly answer, when you can just switch to toml.  There's even sql query engines for toml if you are married to SQL.

Arksine, 2024-10-12:
To be clear sqlite isn't the reason Moonraker will drop support for Python 3.7 and 3.8.  I made that comment to the OP because his log shows he is running 3.7.   Python 3.7 and Python 3.8 have both officially reached EOL status, so they will no longer receive security updates.  Some of Moonraker's  dependencies, such as Tornado, will almost certainly drop support for 3.8 in the near future.  These dependencies often release updates with their own security patches.  In addition, some dependencies may require the latest version to run on the latest stable python release.   Juggling dependencies per version of Python is a maintenance burden I would prefer to drop.

If these systems are air-gapped, presumably they aren't taking Moonraker updates either.   Presumably there should be no issue sticking with the latest Moonraker release that supports Python 3.8 in your situation.

Afinogen, 2025-02-07:
I managed to solve the problem. I have a single-board computer based on ubuntu 18.04. I installed python3.12 and sqlite 3.45 in a separate folder. I followed the example of the article below. You need to specify all the paths correctly, then the system will assemble.

https://www.bluebill.net/install-python-and-sqlite-from-source.html