Arksine/moonraker #865: Missing Documentation for altering the instance_uuid

KlausHans, 2024-05-28:
### What happened

I had a problem with Moonraker, i think after an update, it could not start. After some time i couldn't solve it so i uninstalled it and installed it again via kiauh.
Now it works, but i get a warning:
"Database: Stored Instance ID does not match current Instance ID. Make sure that this database is not shared between multiple Instances of Moonraker."
[image]
I searched and searched but it seems i am the only one with that warning on the internet. Or the solution is buried in Discord.

Could you please add a documentation for how the instance_uuid can be altered?

### Client

Fluidd

### Browser

Firefox

### How to reproduce

- (i think this has nothing to do with it) have a moonraker installation with "moonraker.confighelper.ConfigError: Section [server], option 'klippy_uds_address': Failed to load 'template' component." error
- reinstall moonraker via kiauh
- warning appears

Arksine, 2024-05-29:
This information is more relevant for troubleshooting, so I pushed a commit that no longer warns and logs this condition instead.

When Moonraker first starts it generates a unique ID for the instance, stored in the data folder.  By default the file location is `~/printer_data/.moonraker.uuid`.  If you open this file it will look something like the following:
```
905c9c547b294d8aaa5806756fa93841
```

It appears then when you reinstalled Moonraker this file was deleted and a new UUID was created.  However the database was not deleted, so it retained the previous UUID.    If you want to make the UUID's match, the solution is to login via SSH, open the file with a text editor, then replace the UUID with the `stored uuid` reported in the warning.

KlausHans, 2024-05-29:
I completely forgot to mention that i also executed `./scripts/data-path-fix.sh` because after the reinstall moonraker didn't found the config because it looked in the "wrong" directory. Maybe that script messed with the id.

Anyway, thank you very much for the clarification and the fix. And of course for your work on moonraker.

KlausHans, 2024-05-29:
I can confirm that the warning disappears after i overwrote the current UUID with the stored UUID in `~/printer_data/.moonraker.uuid` and restarted moonraker via `sudo systemctl restart moonraker`.

closed 2024-05-29 by KlausHans