Arksine/moonraker #874: Queue should be paused when non-queued file is finished printing

AndreKR, 2024-06-21:
### What happened

The documentation for the `[job_queue] automatic_transition` setting says:

> When False the queue will be paused after each job is loaded, requiring that users manually resume to load the next print.

This currently doesn't happen if the currently printing file didn't come from the queue.

### Client

Fluidd

### Browser

Chrome

### How to reproduce

- Make sure to have `[job_queue] automatic_transition: False` configured (it's the default)
- Have an empty job queue
- Start a print (the normal way, not by using the queue)
- Add a file to the job queue

Expected result: After the currently printing file is printed, the queue should be paused, so that I can start the next print with <kbd>Resume</kbd>.

Actual result: The queue stays resumed, so in order to print the next file I first have to click <kbd>Pause</kbd> and only then I can print the next file using <kbd>Resume</kbd>.

Arksine, 2024-06-23:
I'm unable to reproduce in my testing using Postman.  When adding an item to an empty queue while Klipper is printing Moonraker's internal queue state is correctly set to `paused` and reflect as such by the `/server/job_queue/status` endpoint.

From your description, it sounds like this may be an issue with the UI/Frontend.  That is, the frontend seems to set its internal state to ready, regardless of the state reported by Moonraker.

AndreKR, 2024-06-29:
Hm, interesting, I can no longer reproduce this, or rather I cannot even resume the queue at all while a print is running, which I guess is the desired behavior? Previously I was able to freely switch between paused and resumed while a print was running.

AndreKR, 2024-06-29:
I was able to reproduce it again. There was an important step missing from my reproduction instructions, the first print has to be started via the queue for the problem to show up.

* Start with an empty queue and no file printing
* Queue state is `paused`
* Add a file to the queue and resume it (`server/job_queue/start`)
* The file starts printing, queue state is now `ready`
* Add another file to the queue (``server/job_queue/post_job`)
* Queue state remains `ready`, it does not switch to `paused` as expected

Arksine, 2024-07-14:
Sorry for the delay.  I was able to reproduce using the method you outlined above.  Commit dc00d38b01915b9aea736999df9287b2846dd6bf should resolve this.

Arksine, 2024-08-09:
As there has been no further feedback, I'm going to close this as resolved.

closed 2024-08-09 by Arksine