Arksine/moonraker #884: APC UPS - Status monitoring

TekuSP, 2024-07-08:
### Is your feature request related to a problem? Please describe

I can't see status of my APC UPS right now.

### Describe the solution you'd like

I would like to have in Moonraker status of APC UPS
- Voltage
- Battery charge
- Remaining time
- Running on backup or mains

And events which could indicate so backup or main was switched (For example for pausing print and cooling down bed to lower temp to survive UPS time, and then when power returns, heat up and continue print)

TekuSP, 2024-07-08:
There is APC Daemon for Raspberry Pi and alternatives: https://gist.github.com/Suzhou65/efc948c5341738d5eab7661cb2dc747e

yell3D, 2024-07-08:
voting for multi vendor implementation 

https://packages.debian.org/search?keywords=nut

https://networkupstools.org/stable-hcl.html

ekorolkov, 2024-09-14:
+1 for NUT support