Arksine/moonraker #890: Cannot start Moonraker - PackageMetadata

jkeveren, 2024-07-25:
### What happened

When starting the moonraker service it reports `ImportError: cannot import name 'PackageMetadata' from 'importlib.metadata' (/usr/lib/python3.9/importlib/metadata.py)` from ` File "/home/biqu/moonraker/moonraker/moonraker.py", line 16, in <module>`

Full output:
```
Jul 25 17:55:16 ultimaker-2 systemd[1]: Started API Server for Klipper SV1.
Jul 25 17:55:16 ultimaker-2 python[7889]: Traceback (most recent call last):
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "/home/biqu/moonraker/moonraker/moonraker.py", line 16, in <module>
Jul 25 17:55:16 ultimaker-2 python[7889]:     svr = importlib.import_module(".server", "moonraker")
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
Jul 25 17:55:16 ultimaker-2 python[7889]:     return _bootstrap._gcd_import(name[level:], package, level)
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "<frozen importlib._bootstrap_external>", line 790, in exec_module
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "/home/biqu/moonraker/moonraker/server.py", line 22, in <module>
Jul 25 17:55:16 ultimaker-2 python[7889]:     from . import confighelper
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "/home/biqu/moonraker/moonraker/confighelper.py", line 17, in <module>
Jul 25 17:55:16 ultimaker-2 python[7889]:     from .utils import Sentinel
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "/home/biqu/moonraker/moonraker/utils/__init__.py", line 25, in <module>
Jul 25 17:55:16 ultimaker-2 python[7889]:     from . import source_info
Jul 25 17:55:16 ultimaker-2 python[7889]:   File "/home/biqu/moonraker/moonraker/utils/source_info.py", line 19, in <module>
Jul 25 17:55:16 ultimaker-2 python[7889]:     from importlib.metadata import Distribution, PathDistribution, PackageMetadata
Jul 25 17:55:16 ultimaker-2 python[7889]: ImportError: cannot import name 'PackageMetadata' from 'importlib.metadata' (/usr/lib/python3.9/importlib/metadata.py)
Jul 25 17:55:17 ultimaker-2 systemd[1]: moonraker.service: Main process exited, code=exited, status=1/FAILURE
```

Ignore the "ultimaker-2" hostname, it's just using the mechanics of an ultimaker 2.

I've resolved this on my system by checking out the parent to a101815 which is where line 19 in `moonraker/moonraker/utils/source_info.py` comes from.

I have run `install-moonraker.sh` after every update and downgrade and also tried re-cloning the entire repo.

I'm using a BIGTREETECH CB1 with mainsail.

### Client

Mainsail

### Browser

Other or N/A

### How to reproduce

Clone the repo, run install-moonraker.sh, check output with journalctl

Given how basic it is to reproduce I imagine this bug is not present on the system that @Arksine is using so here is my setup:
Printer board: BIGTREETECH M8P
Linux host system: BIGTREETECH CB1
Linux Distro: BIGTREETECH CB1 image https://github.com/bigtreetech/CB1 installed around a year ago but updated today.

### Additional information

moonraker.log doesn't seem to have any lines logged at a time that would correspond to the service exit time as shown in the systemd logs. Probably because python doesn't start the program at all. I could be wrong though so I can upload moonraker.log upon request but it's pretty long.

Arksine, 2024-07-25:
This should be resolved by commit 8debbf8ba477064b77e16be7c0d9d3d3c2be1c07.

jkeveren, 2024-07-25:
Haha you pushed the fix as I was writing the issue. Nice one, working perfectly now. Thanks!

closed 2024-07-25 by jkeveren