Arksine/moonraker #892: Print file with '#' in name

sharpner, 2024-07-31:
### What happened

Uploading a file with # in the filename works, but it does not get a preview image, see the difference between the files:

[image]

Trying to print it anyways results in:
[image]

After some testing I figured out that when the filename of the gcode contains the # special character, it won't work.

### Client

Mainsail

### Browser

Safari

### How to reproduce

Upload a gcode file with # in the name. Can by via web interface or directly onto the host (I did it via samba share) does not matter.

### Additional information

mainsail
v2.12.0
moonraker
v0.9.2-0-gb40751ba

Arksine, 2024-07-31:
Hi.  It appears you have two different issues, however I don't believe that either is specific to a bug in Moonraker.

With regard to the missing image, I can only speculate, but I suspect the frontend is not properly escaping the `#` when requesting the image.

The larger issue, which likely makes the image problem moot, is that Klipper cannot accept a `#` as part of a gcode command parameter.  Thus, when Moonraker attempts to run `SDCARD_PRINT_FILE FILENAME=...`, you get the malformed command error.

sharpner, 2024-07-31:
Hey, thanks for the answer!

Makes sense, I would already be happy if the frontend would let me know that # is not allowed when uploading an image - took me too long to find the error

closed 2025-01-20 by Arksine