Arksine/moonraker #893: Metadata extract fails with 'Input data was truncated' some of the time.

Cellivar, 2024-08-03:
### What happened

Most of the time, but not _all_ of the time, Moonraker fails to extract metadata from gcode files. The file prints correctly.

The logs show [this line loading the json](https://github.com/Arksine/moonraker/blob/b40751ba9d7f05e7da81636e3e4a5735ef85bb27/moonraker/components/file_manager/file_manager.py#L2562) as the problem. It doesn't look like there's a lot of logging right around there so I'm uncertain if more verbose logging will help.

This is the error I see in my logs related to metadata parsing:

```
[shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is enabled

[shell_command.py:pipe_data_received()] - INFO:metadata:File 'test_file.gcode' currently supports cancellation, processing aborted

[shell_command.py:_check_proc_success()] - Command (/home/moonraker/venv/bin/python /home/moonraker/moonraker/moonraker/components/file_manager/metadata.py -p /home/moonraker/printer_data/gcodes -f "test_file.gcode" --check-objects) successfully finished

[file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):

  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2520, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)

  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2564, in _run_extract_metadata
    decoded_resp: Dict[str, Any] = jsonw.loads(result.strip())
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
msgspec.DecodeError: Input data was truncated
```
<details>
  <summary>Additional context logs, though not particularly helpful</summary>
  
```
[application.py:prepare()] - Upload Request Received from 192.168.1.44
Content-Type: 
[application.py:prepare()] - Upload Request Received from 192.168.1.44
Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryFov2ibiAVBTX5PEA
[application.py:post()] - Processing Uploaded File: test_file.gcode
[shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is enabled

[shell_command.py:pipe_data_received()] - INFO:metadata:File 'test_file.gcode' currently supports cancellation, processing aborted

[shell_command.py:_check_proc_success()] - Command (/home/moonraker/venv/bin/python /home/moonraker/moonraker/moonraker/components/file_manager/metadata.py -p /home/moonraker/printer_data/gcodes -f "test_file.gcode" --check-objects) successfully finished
[file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2520, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2564, in _run_extract_metadata
    decoded_resp: Dict[str, Any] = jsonw.loads(result.strip())
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
msgspec.DecodeError: Input data was truncated
[shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is enabled

[shell_command.py:pipe_data_received()] - INFO:metadata:File 'test_file.gcode' currently supports cancellation, processing aborted

[shell_command.py:_check_proc_success()] - Command (/home/moonraker/venv/bin/python /home/moonraker/moonraker/moonraker/components/file_manager/metadata.py -p /home/moonraker/printer_data/gcodes -f "test_file.gcode" --check-objects) successfully finished
[file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2520, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2564, in _run_extract_metadata
    decoded_resp: Dict[str, Any] = jsonw.loads(result.strip())
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
msgspec.DecodeError: Input data was truncated
[shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is enabled

[shell_command.py:pipe_data_received()] - INFO:metadata:File 'test_file.gcode' currently supports cancellation, processing aborted

[shell_command.py:_check_proc_success()] - Command (/home/moonraker/venv/bin/python /home/moonraker/moonraker/moonraker/components/file_manager/metadata.py -p /home/moonraker/printer_data/gcodes -f "test_file.gcode" --check-objects) successfully finished
[file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2520, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
[... 34 lines cut ...]
* If I download a correctly extracted file from Fluidd's interface and re-upload it with a different name the extraction may fail.
* If I [use the trick from the missing inotify issue](https://github.com/Arksine/moonraker/issues/614#issuecomment-1435667748) to kick the file Moonraker correctly detects the modified file, kicks off a new metadata extract, and fails.

This is the file I'm using for testing. It has printed successfully several times today.

[test_file.zip](https://github.com/user-attachments/files/16478101/test_file.zip)

To test a bit more I connected to Moonraker and ran the command from the logs directly. The output appears to be well formed JSON:

```
moonraker@72023bea01ef:~/printer_data/gcodes$ /home/moonraker/venv/bin/python /home/moonraker/moonraker/moonraker/components/file_manager/metadata.py -p /home/moonraker/printer_data/gcodes -f "test_file.gcode" --check-objects
INFO:metadata:Object Processing is enabled
INFO:metadata:File 'test_file.gcode' currently supports cancellation, processing aborted
{"file": "test_file.gcode", "metadata": {"size": 8148280, "modified": 1722658371.0097582, "uuid": "5b6f3d29-4b66-459d-82c4-70065558a3a7", "slicer": "OrcaSlicer", "slicer_version": "2.1.1", "gcode_start_byte": 31185, "gcode_end_byte": 8132452, "layer_count": 98, "object_height": 19.6, "estimated_time": 10052, "nozzle_diameter": 0.4, "layer_height": 0.2, "first_layer_height": 0.2, "first_layer_extr_temp": 220.0, "first_layer_bed_temp": 60.0, "chamber_temp": 0.0, "filament_name": "Generic PLA", "filament_type": "PLA", "filament_total": 25679.95, "filament_weight_total": 76.59, "thumbnails": [{"width": 32, "height": 32, "size": 1495, "relative_path": ".thumbs/test_file-32x32.png"}, {"width": 48, "height": 48, "size": 1348, "relative_path": ".thumbs/test_file-48x48.png"}, {"width": 300, "height": 300, "size": 10264, "relative_path": ".thumbs/test_file-300x300.png"}]}}
```

### Additional information

I run my printer software stack [in Nomad](https://www.nomadproject.io/) using Docker. I construct a custom Docker image [using this dockerfile](https://github.com/Cellivar/3d-printing/blob/main/images/moonraker/Dockerfile) which is deployed into my cluster [with these docker parameters](https://github.com/Cellivar/3d-printing/blob/main/terraform/modules/3d_printer/printer_jobs.nomad.hcl#L106-L138) and [this moonraker config file](https://github.com/Cellivar/3d-printing/blob/main/configs/klipper/common/moonraker.conf). The current image I'm running is [based on this commit from July 5th](https://github.com/Arksine/moonraker/commit/dc00d38b01915b9aea736999df9287b2846dd6bf).

My config disables file-based logging as I use a centralized log management system instead. If you would really prefer a file I can generate something close to a moonraker.log for you.

I suspect I'm the only one running Moonraker in the particular system configuration I use, and I am very willing to perform more debugging on my end if you can toss me some pointers. Python is not my forte so I will need a bit of guidance to help me help you :)

Thank you for all of your work on this software package, it's made my printing experience delightful.

Arksine, 2024-08-03:
Hi.  Can you reproduce with verbose logging enabled?  Set the following environment variable:

```
MOONRAKER_VERBOSE_LOGGING="y"
```
This will print out the actual data its attempting to decode, along with any potential errors during metadata extraction. 

You could also try disabling `msgspec` and/or `uvloop` by modifying the environment.
```
MOONRAKER_ENABLE_MSGSPEC="n"
MOONRAKER_ENABLE_UVLOOP="n"
```

Cellivar, 2024-08-03:
I would tell you when my print is going to finish and I can try those, but because of the failure I can't quite tell :D

I should be able to try those later today and report back.

Cellivar, 2024-08-04:
Well that certainly explains the json error, it's _blank_.

```
...
[shell_command.py:_check_proc_success()] - Command (/home/moonraker/venv/bin/python /home/moonraker/moonraker/moonraker/components/file_manager/metadata.py -p /home/moonraker/printer_data/gcodes -f "test_file.gcode" --check-objects) successfully finished

[file_manager.py:_run_extract_metadata()] - Invalid metadata response:
bytearray(b'')

[file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
...
```

<details><summary>The whole upload exchange</summary>

```

[application.py:log_request()] - 200 GET / (192.168.1.196) [No User] 5.75ms
[authorization.py:check_cors()] - CORS Pattern Matched, origin: https://fluidd.squeak.house  | pattern: .*\.squeak\.house
[application.py:prepare()] - Upload Request Received from 192.168.1.57
Content-Type: 
[application.py:log_request()] - 204 OPTIONS /server/files/upload (192.168.1.57) [No User] 4.64ms
[authorization.py:check_cors()] - CORS Pattern Matched, origin: https://fluidd.squeak.house  | pattern: .*\.squeak\.house
[application.py:prepare()] - Upload Request Received from 192.168.1.57
Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryVHdN78TOIeI8xoSO
[application.py:log_request()] - 200 GET / (192.168.1.196) [No User] 5.41ms
[application.py:post()] - 
File Upload Arguments:
root: gcodes
filename: test_file.gcode
tmp_file_path: /tmp/moonraker.upload-2243781.mru
Checksum: 0ad94342816d69a04d0fed80c270f97d1d898d1f4c51a586d4832245b6d12f93
[application.py:post()] - Processing Uploaded File: test_file.gcode
[file_manager.py:_process_file_event()] - Inotify writable file closed: /home/moonraker/printer_data/gcodes/test_file.gcode
[shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is enabled

[shell_command.py:pipe_data_received()] - INFO:metadata:File 'test_file.gcode' currently supports cancellation, processing aborted

[file_manager.py:_process_file_event()] - Inotify writable file closed: /home/moonraker/printer_data/gcodes/.thumbs/test_file-48x48.png
[file_manager.py:_process_file_event()] - Inotify writable file closed: /home/moonraker/printer_data/gcodes/.thumbs/test_file-300x300.png
[file_manager.py:_process_file_event()] - Inotify writable file closed: /home/moonraker/printer_data/gcodes/.thumbs/test_file-32x32.png
[shell_command.py:_check_proc_success()] - Command (/home/moonraker/venv/bin/python /home/moonraker/moonraker/moonraker/components/file_manager/metadata.py -p /home/moonraker/printer_data/gcodes -f "test_file.gcode" --check-objects) successfully finished
[file_manager.py:_run_extract_metadata()] - Invalid metadata response:
bytearray(b'')
[file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2520, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2564, in _run_extract_metadata
    decoded_resp: Dict[str, Any] = jsonw.loads(result.strip())
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
msgspec.DecodeError: Input data was truncated
[shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is enabled

[shell_command.py:pipe_data_received()] - INFO:metadata:File 'test_file.gcode' currently supports cancellation, processing aborted

[file_manager.py:_process_file_event()] - Inotify writable file closed: /home/moonraker/printer_data/gcodes/.thumbs/test_file-48x48.png
[file_manager.py:_process_file_event()] - Inotify writable file closed: /home/moonraker/printer_data/gcodes/.thumbs/test_file-300x300.png
[file_manager.py:_process_file_event()] - Inotify writable file closed: /home/moonraker/printer_data/gcodes/.thumbs/test_file-32x32.png
[shell_command.py:_check_proc_success()] - Command (/home/moonraker/venv/bin/python /home/moonraker/moonraker/moonraker/components/file_manager/metadata.py -p /home/moonraker/printer_data/gcodes -f "test_file.gcode" --check-objects) successfully finished
[file_manager.py:_run_extract_metadata()] - Invalid metadata response:
bytearray(b'')
[file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2520, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 2564, in _run_extract_metadata
    decoded_resp: Dict[str, Any] = jsonw.loads(result.strip())
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 31 lines cut ...]
[application.py:post()] - Processing Uploaded File: test_file.gcode
[file_manager.py:_process_file_event()] - Inotify writable file closed: /home/moonraker/printer_data/gcodes/test_file.gcode
[shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is enabled

[shell_command.py:pipe_data_received()] - INFO:metadata:File 'test_file.gcode' currently supports cancellation, processing aborted

[file_manager.py:_process_file_event()] - Inotify writable file closed: /home/moonraker/printer_data/gcodes/.thumbs/test_file-48x48.png
[... cut ...] _file-300x300.png"}], "print_start_time": null, "job_id": null, "filename": "test_file.gcode"}, "id": 64194}
```

A stress test is in order. `for f in *.gcode; do echo -e "\n" >> "${f}"; done;`

And lo!

[image]

I need to kick off another print and head to bed, so I can't untangle whether msgspec or uvloop was the specific which fixed it. I should have time tomorrow to poke at it again, and I'm happy to perform any other specific debugging steps you think might help narrow down what's going on.

Arksine, 2024-08-04:
My guess is the problem is `uvloop`.  The `msgspec` package is just a replacement python's standard  json encoder/decoder.

The `uvloop` module replaces the built-in asyncio eventloop.  This could impact launching subprocesses.  I don't know if its an issue with Python 3.12 or if there is something about the environment it doesn't like.

Arksine, 2024-08-09:
This appears to be resolved (issue with uvloop), so I am going to close this.

closed 2024-08-09 by Arksine

nelsongraca, 2024-09-09:
@Arksine I'm seeing this error as well, with python 3.12, anything I can do to help troubleshoot this?

Arksine, 2024-09-10:
@nelsongraca Do you have uvloop installed?  Attach `moonraker.log` and I will look into it.

loss-and-quick, 2025-10-06:
@Arksine 

[moonraker.log](https://github.com/user-attachments/files/22730966/moonraker.log)