Arksine/moonraker #897: Failed to Connect to spoolman: [Errno 113] No route to host

JekleFPV, 2024-08-11:
### What happened

Spoolman worked just fine on the latest software about two weeks ago. 

I ran the latest updates available for klipper/moonraker/mainsail/fluidd/spoolman today. Ever since updating, both Mainsail and Fluidd report "Spoolman server not available".

[image]

Not sure where the problem lies, but Moonraker.log reports the following:
Failed to Connect to spoolman: [Errno 113] No route to host

### Client

Mainsail, Fluidd, MobileRaker

### Browser

Safari

### How to reproduce

It was working properly about two weeks ago, I only updated everything to the latest version just today, error has popped up ever since

### Additional information

[moonraker-3.log](https://github.com/user-attachments/files/16574441/moonraker-3.log)

Arksine, 2024-08-11:
The error suggests that the Spoolman server is not available at the configured address/port.  I'd suggest checking the Spoolman service to make sure its running.  I'd also check the IP address and port, if DHCP assigned the host a different IP you would get this error.

JekleFPV, 2024-08-11:
> The error suggests that the Spoolman server is not available at the configured address/port. I'd suggest checking the Spoolman service to make sure its running. I'd also check the IP address and port, if DHCP assigned the host a different IP you would get this error.

Spoolman is running, I can get into its own webUI and interact with it, on the same address and port as before, so none of that should be the issue.

Arksine, 2024-08-11:
For some reason it isn't reachable from Moonraker.  You are experiencing a network error.  Unfortunately I don't have any guidance as to what the best way is to solve it other than my previous suggestions.

JekleFPV, 2024-08-12:
Everything is running on the same RPi, so wouldn't really be a network issue, but is there any file I could check for the localhost address perhaps?

Arksine, 2024-08-12:
> EDIT: never mind, turns out moonraker.conf had a static IP set, instead of localhost.... Smooth sailing again!

Glad you were able to resolve it, I thought it might be a DHCP issue.  I'll close this as complete.

closed 2024-08-12 by Arksine