Arksine/moonraker #909: Using Remote API to call Gcode macro 'A0_TEST' actually calls 'A0' Macro

QwermMakes, 2024-10-06:
### What happened

When you make two gcode macros that both start include a number followed by an underscore (e.g. "A0_FOOBAR" & "A0") and call them using the remote api, calling A0 works as intended, but A0_FOOBAR causes the A0 macro to run.

I've tried a few different versions. Excluding the number before the underscore in the macro's name seems to work normally. For example, calling "BA_TEST" and "BA" both work as intended.

### Client

Mainsail

### Browser

Firefox, Other or N/A

### How to reproduce

Create the following macros:

    [gcode_macro B0]
    gcode:
      M118 B0

    [gcode_macro B0_TEST]
    gcode:
      M118 B0_Test

Use the remote API by calling "http://<ServerURL>/printer/gcode/script?script=B0_TEST

The console will print B0 and not B0_Test.

QwermMakes, 2024-10-06:
I've messed around some more and found out that "B0", "B0_Test", and "B0Test" all call B0, even when inputted directly into the console. Seems to not be an API issue. Maybe just an issue with Klipper's G-Code macros?

Arksine, 2024-10-06:
Correct, this is not a Moonraker issue, nor is it a bug in Klipper.  Its a limitation of extended gcode commands, they cannot begin with something that looks like a standard gcode command (ie: A letter followed by a number).  So it cant be `B0_TEST`, but `TEST_B0` would be fine.

closed 2025-01-20 by Arksine