Arksine/moonraker #915: Moonraker Warning: file_manager: Inotify watch already exists for path '/home/mks/printer_data/config/KAMP/Configuration' in root 'config', cannot add watch to requested root 'config'. This indicates that the roots overlap.

yardi83, 2024-11-05:
### What happened

Hi!
I've been getting this warning since I installed KAMP and I don't know how to get rid of it. I've already googled a bit but the results didn't help me.
I use the latest Moonraker version, installed via KIAUH.
However, Klipper does not run on a Raspberry PI but rather on an MKS PI controller board

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Just installed KAMP

yardi83, 2024-11-09:
so after cleaning up my config_backup folder, delete all KAMP folders and config files and re-installing moonraker via KIAUH i was able to install KAMP again without running into this warning

closed 2024-11-09 by yardi83

RS250R, 2025-01-22:
Got the same error message.  Somehow a shortcut for the KAMP "Configuration" folder ended up in the KAMP Configuration folder.  Guess the circular logic was too much for it to bear...  I deleted the shortcut and all is well.