Arksine/moonraker #917: Moonraker won't reconnect (due to race while server is restarting?)

justingrant, 2024-11-12:
### What happened

I uploaded a printer config file with an error in it. The server therefore failed to start, and the connection to Klippy was closed. But then I reverted the change, but Moonraker didn't re-establlish a connection to Kilppy after the correct file was uploaded. So now Fluidd is reporting (correctly) that Kilppy is disconnected.

Inside [Moonraker's KlippyConnection._on_connection_closed](https://github.com/Arksine/moonraker/blob/4e00a0760e756872b2fbb23a82b88dbceb383cf4/moonraker/components/klippy_connection.py#L765)‎, I saw this:

https://github.com/Arksine/moonraker/blob/4e00a0760e756872b2fbb23a82b88dbceb383cf4/moonraker/components/klippy_connection.py#L780-L784

Could there be a race where if the server happens to be down (because it's restarting) when the Kilppy connection is closed, then Moonraker won't restart the connection later?  If yes, then would this explain the behavior we're seeing?

Here's the Moonraker logs from the first upload of the bad file:

```
2024-11-11 16:05:44,101 [application.py:prepare()] - Upload Request Received from 100.70.249.30
Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryHeoeUIBA2uk2stIk
2024-11-11 16:05:44,138 [application.py:post()] - Processing Uploaded File: /macros/macros_generic.cfg
2024-11-11 16:05:44,143 [application.py:log_request()] - 201 POST /server/files/upload (100.70.249.30) [_TRUSTED_USER_] 42.73ms
2024-11-11 16:05:45,278 [klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed
2024-11-11 16:05:45,281 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: 503, Message: Klippy Host not connected
2024-11-11 16:05:45,305 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 127.0.0.1
2024-11-11 16:05:45,341 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: 503, Message: Klippy Host not connected
2024-11-11 16:05:45,355 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: 503, Message: Klippy Host not connected
2024-11-11 16:05:45,360 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: 503, Message: Klippy Host not connected
2024-11-11 16:05:45,369 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: 503, Message: Klippy Host not connected
```

And here's the log lines from when the fixed file was uploaded:

```
2024-11-11 16:07:59,037 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: 503, Message: Klippy Host not connected
2024-11-11 16:07:59,054 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: 503, Message: Klippy Host not connected
2024-11-11 16:07:59,161 [application.py:prepare()] - Upload Request Received from 100.70.249.30
Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryvu3DlnGAjqRb3FxJ
2024-11-11 16:07:59,191 [application.py:post()] - Processing Uploaded File: /macros/macros_generic.cfg
2024-11-11 16:07:59,197 [application.py:log_request()] - 201 POST /server/files/upload (100.70.249.30) [_TRUSTED_USER_] 35.62ms
2024-11-11 16:07:59,203 [application.py:log_request()] - 503 GET /printer/objects/query?idle_timeout&gcode_move&toolhead&display_status&mcu&heaters&system_stats&fan&heater_bed&heater_generic%20chamber&heater_generic%20chamber1&extruder&extruder1&print_stats (100.121.62.59) [_TRUSTED_USER_] 2.50ms
2024-11-11 16:07:59,267 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: 503, Message: Klippy Host not connected
2024-11-11 16:07:59,269 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.firmware_restart, Code: 503, Message: Klippy Host not connected
```

### Client

Fluidd

### Browser

Chrome

### How to reproduce

1. Upload a config file that causes startup to fail (don't use any backup config)
2. Revert to a known-good version of that file

Expected: Klippy is reconnected
Actual: Klippy is still disconnected.

### Additional information

If this is expected behavior, then what's the recommended recovery step to force the Klippy connection to be re-established? I'd prefer to avoid having to reboot the printer if we can do something that gets the printer back to functional quicker.

Arksine, 2024-11-12:
Please include full logs, snippets don't provide all the information required diagnose a potential issue.

>  Moonraker didn't re-establlish a connection to Kilppy after the correct file was uploaded.

Merely uploading a new configuration file wouldn't be enough to bring Klippy back up in the event of a crash.  You would need to restart the Klipper service.

> Could there be a race where if the server happens to be down (because I broke it by the bad config file) when the Kilppy connection is closed, then Moonraker won't restart the connection later after Klippper is healthy? 

No.  The server does not depend on Klippy.  This line of code prevents Moonraker from attempting to reconnect after the server is commanded to exit.

Arksine, 2025-01-20:
As there has been no further discussion I am going to close this.

closed 2025-01-20 by Arksine