Arksine/moonraker #918: system dependencies not compatible with Ubuntu 24.10

don17sch, 2024-11-16:
### What happened

Ubuntu 24.10 replaced the wireless-tools package with the iw package. 
in order to get moonraker installed I have to manually install the iw package and manually delete the wireless-tools package from the system-dependencies.json file

### Client

Other

### Browser

Other or N/A

### How to reproduce

install ubuntu server 24.10
git clone https://github.com/Arksine/moonraker.git
./moonraker/scripts/install-moonraker.sh
see failure
sudo apt update && sudo apt install iw
nano ~/moonraker/scripts/system-dependencies.json
delete line with text "wireless-tools",
save file
runs install moonraker script again

Arksine, 2024-11-22:
I'm currently looking into this.  It will require that Moonraker's `system dependencies` implement requirement specifiers similar to what is available for Python.  I don't see this as a critical issue at the moment, as I'm more focused on supporting Debian and LTS Ubuntu releases, however I do think its a good idea to get this done now since its likely that a growing number of users will upgrade when 25.04 is released.

Arksine, 2025-01-20:
This should now be resolved.

closed 2025-01-20 by Arksine

theo-goslett, 2025-02-22:
I am still facing the same bug, when I attempt to install via KIAUH I get the following:

###### Installing dependencies ...

###### Reading dependencies...
python3-virtualenv
python3-dev
libopenjp2-7
libsodium-dev
zlib1g-dev
libjpeg-dev
packagekit
wireless-tools;distro_id!='ubuntu'ordistro_version<='24.04'
iw;distro_id=='ubuntu'anddistro_version>='24.10'
curl
build-essential

###### Installing packages...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package wireless-tools;distro_id!='ubuntu'ordistro_version<
E: Couldn't find any package by glob 'wireless-tools;distro_id!='ubuntu'ordistro_version<'
E: Unable to locate package iw;distro_id=='ubuntu'anddistro_version>
>>>>>> Installing Moonraker packages failed!

I can bypass this using the fix provided by don17sch but this then leaves me with an improperly configured instance of Moonraker and I cannot access Fluidd. As a beginner to the ecosystem this is challenging to resolve.

Arksine, 2025-02-22:
This is an issue with KIAUH.  The developer is currently working on resolving it.

Arksine, 2025-02-22:
Update: The latest version of KIAUH should now be able to install Moonraker

dezigenb, 2025-04-26:
Hmm, I think KIAUH didn't fix this issue. As you can see, after your merge, in the moonraker.sh, the system_deps still have the package version restriction. In my opinion, the restriction is originate from  moonraker/scripts/system-dependencies.json. In fact, after my test, removing those two restrictions doesn't cause any problems. And it can work perfectly on all devices.

[image]