Arksine/moonraker #919: "[No User]" error when signing in with LDAP

justinhschaaf, 2024-11-19:
### What happened

Hello! I've gone ahead and configured authentication with LDAP as described in the documentation for [Moonraker](https://moonraker.readthedocs.io/en/latest/configuration/#ldap) and [Fluidd](https://docs.fluidd.xyz/authorization#ldap---configuration). Now, whenever I try to sign in using my LDAP credentials, Fluidd shows an "Invalid credentials" error message, and Moonraker's logs show an error 401 with the message "[No User]". I've manually updated Fluidd, Moonraker, and Klipper to the latest versions as described in [Fluidd's documentation](https://docs.fluidd.xyz/updates/manual) to no avail.

The LDAP server I'm trying to authenticate through is [Authentik's LDAP outpost](https://docs.goauthentik.io/docs/add-secure-apps/providers/ldap/), which I've verified is working properly through manual queries with the ldapsearch command and by configuring it with another application (OPNsense). I can also see Moonraker properly querying the server in its logs.

### Client

Fluidd

### Browser

Chrome, Firefox

### How to reproduce

Assuming you already have working Moonraker, Fluidd, and Authentik instances:

1. [Setup an LDAP in provider as described here.](https://docs.goauthentik.io/docs/add-secure-apps/providers/ldap/generic_setup)
    - I personally use Authentik with [authentik-nix](https://github.com/nix-community/authentik-nix), which you can view my full configuration for [here](https://github.com/justinhschaaf/nixos-config/blob/main/modules/server/services/authentik.nix). The environment file for authentik-ldap has the `AUTHENTIK_HOST`, `AUTHENTIK_TOKEN`, and `AUTHENTIK_INSECURE` variables provided when viewing the outpost in Authentik.
2. Update `moonraker.conf`, adding a `[ldap]` section pointing to the Authentik outpost and updating `[authorization]` to force logins and default to LDAP.
    - The user filter is optional. This should make it so users must have the `iam-fluidd` group to authenticate. The issue still occurs without this being declared.

    <details>

    ```ini
    [ldap]
    ldap_host: [LDAP IP ADDRESS]
    ldap_port: 389
    ldap_secure: False
    base_dn: dc=ldap,dc=goauthentik,dc=io
    bind_dn: [SERVICE ACCOUNT USERNAME]
    bind_password: [SERVICE ACCOUNT PASSWORD]
    user_filter: (&(objectClass=user)(cn=USERNAME)(memberOf=cn=iam-fluidd,ou=groups,dc=ldap,dc=goauthentik,dc=io))
    
    [authorization]
    force_logins: True
    default_source: ldap
    cors_domains:
    ...
    ```

    </details>

3. Restart Moonraker with `systemctl restart moonraker`. Navigate to your Fluidd instance in your web browser of choice and try signing in using your Authentik/LDAP account credentials.

### Additional information

[moonraker.log](https://gist.github.com/justinhschaaf/ae1de8842c3362d9b368ef27aa8c2fc2#file-moonraker-log) - The login attempts are at the very bottom of the file.

[moonraker.conf](https://gist.github.com/justinhschaaf/ae1de8842c3362d9b368ef27aa8c2fc2#file-moonraker-conf)

Relevant logs in the LDAP server

<details>

[image]

</details>

Error message as seen in Fluidd

<details>

[image]

</details>

Arksine, 2024-11-22:
Most of the ldap functionality is contributed.  Since I do not use ldap myself I only have a high level understanding of its inner workings.  I'm willing to take a look at this, however it would be useful if you can reproduce the issue with verbose logging enabled.  Presuming you are on a typical installation, this can be done by adding `MOONRAKER_VERBOSE_LOGGING="y"` to  Moonraker's environment file (`<data_folder>/systemd/moonraker.env`).  The [documentation](https://moonraker.readthedocs.io/en/latest/installation/#the-environment-file) provides further detail.

justinhschaaf, 2024-11-23:
[Here's](https://gist.github.com/justinhschaaf/ae1de8842c3362d9b368ef27aa8c2fc2#file-verbose-moonraker-log) the updated log file with verbose logging enabled via the environment variable. I'm definitely not an expert with LDAP either, but I'll do my best to take another close look at this as well when I get some more free time. I really appreciate your help with this, thank you so much!

Arksine, 2024-11-24:
After looking at the log I don't see anything that would suggest a bug in the ldap code, so I suspect there is a configuration issue.  The group DN shouldn't be part of the `user_filter`, it should be set in the [group_dn](https://moonraker.readthedocs.io/en/latest/configuration/#ldap) option.  If it still doesn't work without removing that part I suspect the `user_filter` is missing something.

justinhschaaf, 2024-11-30:
I went ahead and removed the group DN from `user_filter` (using only `(&(objectClass=user)(cn=USERNAME))`, as per the [docs](https://moonraker.readthedocs.io/en/latest/configuration/#ldap)) and kept the `group_dn` option unset, unfortunately still experiencing the issue.

I realized that error messages were only being sent along with the HTTP response instead of to the output logs, so I checked the browser console and saw the error was `LDAP authentication failed`, sent from [here](https://github.com/Arksine/moonraker/blob/master/moonraker/components/ldap.py#L107).

[image]

Adding a quick debug message, the ldap3 module throws the error `invalid attribute type createTimestamp` (I've also seen it mention `modifyTimestamp` instead). Both are missing from the list of fields set for users in [authentik's docs](https://docs.goauthentik.io/docs/add-secure-apps/providers/ldap/). [This is also an issue when using Google's LDAP connector.](https://github.com/cannatag/ldap3/issues/1045)

I didn't get the chance to look into exactly where ldap3 tries to validate it and runs into the issue; however, when inputting an invalid username, it does return [the appropriate error](https://github.com/Arksine/moonraker/blob/master/moonraker/components/ldap.py#L88).

I tried adding `createTimestamp` and `modifyTimestamp` to the list of attributes not checked by ldap3 (see https://github.com/cannatag/ldap3/issues/557#issuecomment-445409554) and [setting `return_empty_attributes` in the connection arguments to False](https://ldap3.readthedocs.io/en/latest/tutorial_searches.html#what-about-empty-attributes), both to no avail. Unless there's a way to have ldap3 not care about these core missing attributes, authentik will probably have to set them for users (or I'll have to see if there's a way to set them manually).

BastelBaus, 2025-01-04:
Same error for me for another application (radicale) with authentik. What was the solution, who did you set the attributes in authentik LDAP provider?

justinhschaaf, 2025-01-17:
> Same error for me for another application (radicale) with authentik. What was the solution, who did you set the attributes in authentik LDAP provider?

Apologies for the slow response, I've been preoccupied with other projects these past couple weeks. I wasn't able to find any clear documentation on setting custom LDAP attributes for all users, As such, I tried [modifying the user](https://docs.goauthentik.io/docs/users-sources/user/user_basic_operations#modify-a-user) and adding the following to the `Attributes` section:

```yaml
createTimestamp: 20250117114153Z
modifyTimestamp: 20250117114153Z
```

I also tried the date/time format described [here](https://unix.stackexchange.com/a/657672); however, I ran into the same exact error both times. I was able to manually perform the LDAP query and saw the attributes were set though:

[image]

I'm absolutely not an expert in LDAP so take my word with a grain of salt, but my guess is that these attributes have an actual date/time type that ldap3 is expecting in addition to the attributes being present in the first place. As the Authentik LDAP Outpost likely can not/does not set the type simply based on what's set as the custom attributes, ldap3 fails in the same way as when they're not set. As per https://github.com/cannatag/ldap3/issues/1017#issuecomment-1839476368, these attributes are mandatory to the schema it expects, and you can't really work around it.

There's an open issue to try and get Authentik to support these attributes (https://github.com/goauthentik/authentik/issues/10474), but there hasn't been much movement on it as of writing.

DanAE111, 2025-01-23:
I've also been having issues that may be related to this issue i am using Active Directory as a backend provider though. I've done some tests with different configuration options enabled and disabled and it seems based on the Moonraker logs the LDAP authentication succeeds.

My ldap config:
```
ldap_host: ad.contoso.com
ldap_secure: true

base_dn: DC=ad,DC=contoso,DC=com

bind_dn: {secrets.ldap.credentials.bind_dn}
bind_password: {secrets.ldap_credentials.bind_password}

group_dn: CN=3D Printer Users,OU=Groups,DC=ad,DC=contoso,DC=com

is_active_directory: true
```

Logs with config as above:
```
2025-01-23 22:47:30,961 [application.py:_log_debug()] - HTTP Request::POST /access/login::{'username': '<sanitized>', 'password': '<sanitized>', 'source': '<sanitized>'}
2025-01-23 22:47:31,929 [ldap.py:_validate_group()] - LDAP User user group match success, login successful
2025-01-23 22:47:31,936 [application.py:log_request()] - 401 POST /access/login (192.168.7.195) [No User] 980.26ms
```

Logs with bad credentials:
```
2025-01-23 22:48:07,865 [application.py:_log_debug()] - HTTP Request::POST /access/login::{'username': '<sanitized>', 'password': '<sanitized>', 'source': '<sanitized>'}
2025-01-23 22:48:08,561 [ldap.py:_perform_ldap_auth()] - LDAP Rebind failed, attempting to validate credentials with new connection.
2025-01-23 22:48:08,883 [application.py:log_request()] - 401 POST /access/login (192.168.7.195) [No User] 1021.59ms
```

Logs with no group filter:
```
2025-01-23 22:49:09,237 [application.py:_log_debug()] - HTTP Request::POST /access/login::{'username': '<sanitized>', 'password': '<sanitized>', 'source': '<sanitized>'}
2025-01-23 22:49:10,265 [ldap.py:_validate_group()] - LDAP User user login successful
2025-01-23 22:49:10,278 [application.py:log_request()] - 401 POST /access/login (192.168.7.195) [No User] 1046.61ms
```

Seems a bit strange that this issue affects three different LDAP providers, something else change somewhere?

Happy to assist with any further testing and provide more logs if required.

DanAE111, 2025-01-31:
@Arksine thanks for the advice you were indeed correct it was because i had an existing local user with the same username i just assumed that because i could select an authentication provider at login the user names clashing wouldn't be an issue.

Possibly worth making a note of that in the docs, would avoid silly people like me making similar mistakes in future.

BastelBaus, 2025-03-23:
Finally, I found a quick&dirty fix for me which worked.

Go inside the container, edit ldap3 config file:

vi /venv/lib/python3.12/site-packages/ldap3/utils/config.py

Line 33ff add to _ATTRIBUTES_EXCLUDED_FROM_CHECK both attributes 'modifyTimestamp', 'createTimestamp'.

Solution is of course not persistant and not a good one. But at least it worked for me ....

BR, BastelBaus

justinhschaaf, 2025-04-12:
@BastelBaus I added the line below (based on [your fix for Radicale](https://github.com/Kozea/Radicale/pull/1740)) to my Moonraker install, specifically after [the `__init__` function of the LDAP module](https://github.com/Arksine/moonraker/blob/7cdcca3cb4b7caf27d511d1c4e32fa3297391709/moonraker/components/ldap.py#L25), and it works flawlessly.

```python
ldap3.utils.config._ATTRIBUTES_EXCLUDED_FROM_CHECK.extend(['createTimestamp', 'modifyTimestamp'])
```

Out of curiosity, I revisited [the previous fix I tried](https://github.com/Arksine/moonraker/issues/919#issuecomment-2508918231) based on https://github.com/cannatag/ldap3/issues/557#issuecomment-445409554 and it appears to have the same effect. Not sure where I went wrong trying this previously, important thing is there's a workaround now.

```python
ldap3.set_config_parameter('ATTRIBUTES_EXCLUDED_FROM_CHECK', ldap3.get_config_parameter('ATTRIBUTES_EXCLUDED_FROM_CHECK') + ['createTimestamp', 'modifyTimestamp'])
```

Thank you so much for figuring this out, I truly appreciate your help!