Arksine/moonraker #924: JSON-RPC Request Error

chrisman1015, 2024-12-06:
### What happened

I'm new to moonraker and having issues connecting to my Ender 5 pro. I've uninstalled and reinstalled klipper, moonraker, and fluidd multiple times. I've checked and rechecked the serial port. Not sure How to fix it. 
 &nbsp;  
 &nbsp;  
klipper.log looks fine (I'm new so idk for sure), but moonraker.log shows it's stuck on a request. I keep getting a repeated error as follows:   

***[common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found***
 &nbsp;  
 &nbsp;  
Any and all help is appreciated. log files: [klipper](https://github.com/chrisman1015/klipper/blob/main/klippy.log), [moonraker](https://github.com/chrisman1015/klipper/blob/main/moonraker.log)

### Client

Fluidd

### Browser

Firefox

### How to reproduce

Install klipper, moonraker, fluidd on raspi zero 2 w

terri1769, 2024-12-07:
Running into same issue on Intel NUC.

closed 2025-01-20 by Arksine

alextac98, 2025-05-30:
Also ran into this issue personally, and I wanted to add my fix to help folks in the future!

My problem was that I had the wrong `klippy_uds_address`, specifying `~/printer_data/comms/klippy.sock` instead of /home/user/printer_data/comms/klippy.sock`. Turns out `~` doesn't get resolved properly!