Arksine/moonraker #927: unable to get moonraker to connect to mainsail

kadukong, 2024-12-15:
### What happened

I am trying to install Klipper with kaiuh and I'm not sure where I've gone wrong. after installing everything I'm trying to go login to mainsail only to be met with a unable to connect to moonraker sign. I've trying to see if Moonraker is active on Putty and it keeps saying is activating. I've tried some different scripts I found online off the Moonraker site and nothing seems to help, I've even tried re-installing just Moonrker and nothing. I'm using a Raspberry Pi 3 and I don't have any idea what I'm doing if there's anyone who can help I'd greatly appreciate it 

### Client

Mainsail

### Browser

Chrome

### How to reproduce

just log into mainsail and I am stuck

### Additional information

[image]

philipphoh, 2025-01-29:
I have exactly the same problem, but on an OrangePi Zero on which Klipper used to run flawlessly before: Mainsail shows a "Cannot connect to Moonraker" error, the moonraker-error.log shows an "Error 111", the service exits on failure with the exact same message as in the provided screenshot. 
This happens when installing via KIAUH v6 alpha, KIAUH v5, as well as when installing manually via `./install-moonraker.sh `. 
Using the [Moonraker python package](https://moonraker.readthedocs.io/en/latest/installation/#installing-the-moonraker-python-package) I got the service to start, but that's not what I want. 
When I find the time, I will try using a previous version of Moonraker to see what happens and report back.

philipphoh, 2025-01-30:
Using a previous release worked. For no specific reason I randomly chose [Commit 63578ae](https://github.com/Arksine/moonraker/commit/63578aefd0ad62b870b79d2ed6d6f475cbf35e03) and after using the install-moonraker.sh everything runs fine. 
The steps I took are the following: 
1. uninstall moonraker completely using KIAUH v6 alpha
2. 
``` bash
cd ~
git clone https://github.com/Arksine/moonraker.git
cd moonraker
git reset --hard 63578ae
git clean -df
./scripts/install-moonraker.sh
```

Schroeder0815, 2025-07-20:
> Using a previous release worked. For no specific reason I randomly chose [Commit 63578ae](https://github.com/Arksine/moonraker/commit/63578aefd0ad62b870b79d2ed6d6f475cbf35e03) and after using the install-moonraker.sh everything runs fine. The steps I took are the following:
> 
> 1. uninstall moonraker completely using KIAUH v6 alpha
> 
> cd ~
> git clone https://github.com/Arksine/moonraker.git
> cd moonraker
> git reset --hard 63578ae
> git clean -df
> ./scripts/install-moonraker.sh
>  3. update moonraker from `v0.9.1-0` to the current version via the Update Manager in Mainsail

Thanks a lot! This way saved me from a completely new installation!!!!

iwant2fly, 2025-08-11:
This worked for me as well.