Arksine/moonraker #931: Update manager won't start after update

plandevida, 2024-12-31:
### What happened

I just updated Moonraker using the update manager in Mainsail and now the update manager won't start.

```
2024-12-31 09:49:30,560 [server.py:load_component()] - Unable to load component: (update_manager)
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/server.py", line 300, in load_component
    module = importlib.import_module(full_name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/pi/moonraker/moonraker/components/update_manager/__init__.py", line 8, in <module>
    from . import update_manager as um
  File "/home/pi/moonraker/moonraker/components/update_manager/update_manager.py", line 19, in <module>
    from .python_deploy import PythonDeploy
ImportError: cannot import name 'PythonDeploy' from 'moonraker.components.update_manager.python_deploy' (/home/pi/moonraker/moonraker/components/update_manager/python_deploy.py)
2024-12-31 09:49:30,561 [server.py:add_warning()] - Unparsed config section [update_manager] detected.  This may be the result of a component that failed to load.  In the future this will result in a startup error.
2024-12-31 09:49:30,561 [server.py:add_warning()] - Unparsed config section [update_manager mainsail] detected.  This may be the result of a component that failed to load.  
In the future this will result in a startup error.
```

[moonraker-3.log](https://github.com/user-attachments/files/18280630/moonraker-3.log)

### Client

Mainsail

### Browser

Safari

### How to reproduce

Open Mainsail and update Moonraker to the last version.

Restart the printer.

The update manager is missing on Mainsail and some warnings about unable to parse update_manager will appear.

plandevida, 2024-12-31:
I just found out the python_deploy.py script was missing during the update.

[image]
[image]

Arksine, 2025-01-01:
The issue is with your local repo.  Note that its been modified to remove `python_deploy.py`.  Since it is unlikely that you did this yourself, it is indicative of File System corruption or OS bugs.

You should be able to fix it by resetting the local repo via SSH:

```
cd ~/moonraker
git reset --hard origin/master
sudo systemctl restart moonraker
```

plandevida, 2025-01-01:
I did reset the repository and all is fine, and all the OS is just working fine, not other issues like this one.

Just wondering if the git command could have caused it. Anyway maybe this does not go anywhere.

closed 2025-01-01 by plandevida