Arksine/moonraker #932: POWER control new type: shell script

Jaj0, 2025-01-01:
### Is your feature request related to a problem? Please describe

[power printer] config has a lot of different types of supported switches, but the issue is that all of them are very device specific.
But there might be cases that all of available solutions are not working for specific case. Then some universal solution would be appropriate.

Example case: im running klipper+moonraker+mainsail on NanoPi Neo Air, my power relay (that shuts down whole printer with MCU) is connected to one of gpios. As this is not RaspberryPI there are some issues to use gpio type for this config.
But i can switch gpio using alternative WiringNP library, just calling gpio write 7 1.

### Describe the solution you'd like

It would be good to add option to have possibility to execute shell command as power on/power off command. Similar as in linked discussion
`[power PRINTER]
type:command
TurnON: gpio mode 7 output; gpio write 7 1
TurnOFF: gpio write 7 0`

### Describe alternatives you've considered

I've tried using shell macros extension for klipper, but issue is that it only works when MCU is already running which is not possible before powering MCU of printer.

Jaj0, 2025-01-01:
https://github.com/Arksine/moonraker/discussions/784

Twinki14, 2025-01-02:
+1, I have a TP Link Tapo smart plug between my printer and power supply, but the plug is no longer natively compatible with Moonraker, this tool https://github.com/python-kasa/python-kasa let's me control them from a CLI - using Klippers shell command extensions I have some level of control on my MCU

The CLI/Library for controlling my smart plug could be wrapped in a HTTP server which I've already made an issue about here https://github.com/python-kasa/python-kasa/issues/1403 but if Moonraker had the ability to just execute the CLI, there would be no need

Jaj0, 2025-01-02:
In the meantime ive created temporary solution in means of simple http server that takes http requests and calls cli commands:
https://github.com/Jaj0/moonraker-cli-switch-server

LittleNyanCat, 2025-02-21:
+1, I'm using an Arduino connected via USB controlling and ATX PSU and a python script to tell it to turn on/off.

Will probably just implement it myself soon and PR it, doesn't look too hard.

mdm351981, 2025-05-03:
+1 I’m using a i2c relay to switch power to printer PSU and being executable via [printer power] would tidy things up a lot.