Arksine/moonraker #936: Missing dbus_fast module

oleksiikutuzov, 2025-01-12:
### What happened

When updating moonraker to the latest via Fluidd today, pip was upgraded from v22 to v24 and was recompiling the dependencies. After the installation and reboot, the moonraker was broken. Checking the logs, the `dbus_fast` module was missing. 
Also weird output in the log (attaching the full log as well)
```
2025-01-13 04:07:40,416 [pip_utils.py:_process_subproc_output()] - Ignoring tornado: markers 'python_version == "3.7"' don't match your environment
2025-01-13 04:07:40,426 [pip_utils.py:_process_subproc_output()] - Ignoring pillow: markers 'python_version == "3.7"' don't match your environment
2025-01-13 04:07:40,439 [pip_utils.py:_process_subproc_output()] - Ignoring streaming-form-data: markers 'python_version == "3.7"' don't match your environment
2025-01-13 04:07:40,451 [pip_utils.py:_process_subproc_output()] - Ignoring dbus-fast: markers 'python_version >= "3.9"' don't match your environment
2025-01-13 04:07:40,468 [pip_utils.py:_process_subproc_output()] - Ignoring importlib_metadata: markers 'python_version == "3.7"' don't match your environment
```

Manual installation of the module into `moonraker-env` fixes the problem, but there seems to be the issue with dependencies.

### Client

Fluidd

### Browser

Other or N/A

### How to reproduce

When troubleshooting, tried installing moonraker fresh via the Creality-Hepler-Script, but it always ended up with the same issue. 

Python and pip versions:
```
(moonraker-env) root@Ender3V3KE-B12A /usr/data/octoapp [#] /usr/data/moonraker/moonraker-env/bin/python3 --version
Python 3.8.2
(moonraker-env) root@Ender3V3KE-B12A /usr/data/octoapp [#] /usr/data/moonraker/moonraker-env/bin/pip3 --version
pip 24.0 from /usr/data/moonraker/moonraker-env/lib/python3.8/site-packages/pip (python 3.8)
```

### Additional information

Printer: Ender 3 V3 KE
Full log: [moonraker.log](https://github.com/user-attachments/files/18391279/moonraker.log)

Guilouz, 2025-01-12:
Can you try this on Ender-3 V3:

- Connect to SSH and run these commands (one at time):

```
/etc/init.d/S56moonraker_service stop
```

```
cd /usr/data/moonraker/moonraker-env/bin
```

```
python3 -m pip install --no-cache-dir "dbus-fast<=2.28.0"
```

```
/etc/init.d/S56moonraker_service start
```

NiXTheDev, 2025-01-12:
> Can you try this on Ender-3 V3:
> 
> * Connect to SSH and run these commands (one at time):
> 
> ```
> /etc/init.d/S56moonraker_service stop
> ```
> 
> ```
> cd /usr/data/moonraker/moonraker-env/bin
> ```
> 
> ```
> python3 -m pip install --no-cache-dir "dbus-fast<=2.28.0"
> ```
> 
> ```
> /etc/init.d/S56moonraker_service start
> ```

So, basically, that is just a problem with how helper-script for klipperized creality printers handles thing by installing old dbus and not the newer drop-in dbus_fast...

Great... gotta wait for that thing to update so that users don't get their moonraker installs just broken
(btw reverting to prev commit also works lol)

ahem, also, please pin the issue

oleksiikutuzov, 2025-01-12:
That was indeed fast, thanks, closing

closed 2025-01-12 by oleksiikutuzov