Arksine/moonraker #940: Moonraker Secrets

primate25, 2025-01-19:
Trying to add secrets in my config but without sucess although I think I made everything like mentioned in the docs and issue below.

Using BTT CB1 on Manta M8P. Moonraker.log attached

[moonraker (8).log](https://github.com/user-attachments/files/18469202/moonraker.8.log)

The secret file is defined in  /home/biqu/printer_data/moonraker.secrets

`{
    "spoolman": {
        "ip": "http://XXX.XXX.XXX.XXX:7912"
    },
    "server": {
        "port": "7125",
        "ssl_port": "7130"
    }
}
`

Any help is appriciated!

### Discussed in https://github.com/Arksine/moonraker/discussions/610

Arksine, 2025-01-20:
Secrets are only available for options that support templates, the configuration documentation will note when this is the case.  Typically this includes options that contain sensitive data, such as user credentials and tokens.

primate25, 2025-01-20:
OK, I understand now and thank you for the clarification although I think IP addresses are also sensitive information that I maybe do not want to share. 

All in all thanks.

closed 2025-01-20 by primate25