Arksine/moonraker #949: Documentation: what are the possibile values of printer state?

beelsebob, 2025-02-03:
### What happened

The documentation for /server/info, and /printer/info specify a "state" property, but they do not specify what values might occur, other than showing the "ready" state.  What other values are possible here?

### Client

Other

### Browser

Other or N/A

### How to reproduce

Look at the docs.

pedrolamas, 2025-02-07:
For `server/info`, the current values are defined [here](https://github.com/Arksine/moonraker/blob/71ab2fb80e3e24f22219ff9358d595e97b5a40d6/moonraker/common.py#L137) (they are shown in uppercase in python, but will be lowercase on the JSON response)

For `printer/info`, those are provided by Klipper and currently are the `category` values you can see [here](https://github.com/Klipper3d/klipper/blob/fec3e685c92ef263a829a73510c74245d7772c03/klippy/klippy.py#L45-L54).

beelsebob, 2025-02-07:
Yup, I found them, the bug is not a question - it's a comment that the documentation is not very good at explaining what's expected here.

Arksine, 2025-02-07:
FWIW, I pushed a very large rework of the documentation yesterday.  Included are full response specifications. 

Go to the [server.info](https://moonraker.readthedocs.io/en/latest/external_api/server/#query-server-info) endpoint, view the response specification, and click on `klippy_state` to expand and view the states available.  The server endpoint reports an extra `disconnected` state when Moonraker is not connected to Klippy.

beelsebob, 2025-02-07:
Oh wow, that's an awesome change!  Thanks so much!