Arksine/moonraker #951: Internal Server Error when trying to upload file via QT5

Aceoflaub, 2025-02-03:
### What happened

First of all, English is not my first language, so sorry for possible mistakes.

I'm trying to write my own simple client to communicate with moonraker. I do it using qt 5.12.12 c++. I've already done JSON-RPC requests and HTTP GET requests but stuck with HTTP POST requests to upload file. When I try to upload a file a get: "Error transferring http://klippi.local/server/files/upload - server replied: Internal Server Error" (klippi is a name of my raspberry pi). I can upload a file via Mainsail web client. I can upload a file via python script. But I get error when I try to do it via qt c++.

### Client

Other

### Browser

Other or N/A

### How to reproduce

Just run a qt5 project that I attach.

### Additional information

moonraker.log says: [application.py:post()] - Processing Uploaded File: None. So I think something went wrong while file transferring. I use wireshark to see packets and get next:
When mainsailweb create post request, it create single tcp segment.
When qt5 create post request, it create two tcp segments separated by single boundary(see attachment qt5postrequest.png).
When python create post request, it create two tcp segments separated by double boundary(see attachment pythonpostrequest.png).
In all other fields requests are the same(python omits optional "root" and "path" arguments, but I try omit this in qt and nothing changes). So I think that maybe moonraker can't properly resolve multiple tcp segments when they separated by single boundary?

In scripts.zip attachment I put files of my qt5 project and python script.

[moonraker.log](https://github.com/user-attachments/files/18640073/moonraker.log)

[scripts.zip](https://github.com/user-attachments/files/18640369/scripts.zip)

[image]

[image]

[image]

Arksine, 2025-02-03:
You are running a version of Moonraker that is out of date, pull the latest version.  Make sure you upgrade the python requirements as well, ie:
```
~/moonraker-env/bin/pip install -U -r ~/moonraker/scripts/moonraker-requirements.txt
```

That said, I'm not sure if that will correct the issue.  Moonraker depends on the [streaming-form-data](https://github.com/siddhantgoel/streaming-form-data) package to parse mulitpart uploads, and it appears its unable to parse the file name.  It very well could have something to do with the way QT is sending the request.

Aceoflaub, 2025-02-04:
> You are running a version of Moonraker that is out of date, pull the latest version.<

That's help me!!! The latest version of Moonraker contains more detailed error messages in the log file. It my case it was: "Multipart filename omitted". It's all my own mistake: as for standard, multiple parameters in Content-Disposition header must be separated by semicolon. And in one parameter I mistakenly typed a comma instead. I fixed this comma to a semicolon and everything worked.

Thank you so much! If it wasn't for your answer, it would take me a while to find this error.

Arksine, 2025-02-05:
Great!  Glad you were able to solve the issue.

closed 2025-02-05 by Arksine