Arksine/moonraker #953: Too strict pdb version dependency

diggit, 2025-02-04:
### Is your feature request related to a problem? Please describe

Hi,
is is really necessary to depend exactly on [pdm 2.3.3](https://github.com/Arksine/moonraker/blob/f3b0afd57491478b259f1d1b3809b4fb71fdc226/pyproject.toml#L75)? It is then problematic to build and install in environments (e.g. system wide in archlinux) with more up to date pdm (currently 2.22.x).

### Describe the solution you'd like

Make the version requirement as a minimum or remove it entirely.

Arksine, 2025-02-04:
The pdm developer has a habit of making breaking changes to the build extensions, so I have to pin the dependency.  I'll look into updating to the latest version.

Arksine, 2025-02-04:
Just checked and the build works with the latest release, 2.4.3, so I bumped the version.

Arksine, 2025-02-05:
This should be resolved, so I am going to go ahead and close this.

closed 2025-02-05 by Arksine