Arksine/moonraker #954: MQTT not publishing some webhook status changes

eagle1361, 2025-02-06:
### What happened

When webhooks updates the status to "startup" or "ready", as shown by /printer/objects/query?webhooks, the mqtt client does not publish either of those status changes. It does update when the status is "shutdown". Changing enable_moonraker_api or publish_split_status does not change behavior.

Section:
[mqtt]
address: mqttserveraddress
enable_moonraker_api: True
instance_name: klipper
publish_split_status: True
status_objects:
  webhooks
  print_stats=state,message

### Client

Other

### Browser

Other or N/A

### How to reproduce

Set MQTT to publish status_object: webhooks. Monitor via MQTT server and web browser get query function. startup and ready status are shown in web browser but not on MQTT server. Shutdown klipper and shutdown message is shown on both locations as expected.

### Additional information

[moonraker.log](https://github.com/user-attachments/files/18698552/moonraker.log)

Arksine, 2025-02-07:
This is due to the nature of how Moonraker initializes its connection to Klippy.  It polls Klipper's `info` endpoint until Klipper is out of the startup phase before it requests subscriptions.  One reason for this is timing, the internal webhooks state may already be beyond the `startup` phase by the time Moonraker can establish a connection, so using a  `webhooks` subscription isn't a reliable way to detect the the `startup` or `ready` state.

That said, I can look into publishing a status update after the initial subscription, so at the very least you would get the `ready` state.

Arksine, 2025-02-21:
FWIW, this should be resolved.  The `startup` state can't be pushed because its invalid to subscribe before Klipper's startup is complete, but `ready` should be.

closed 2025-02-21 by Arksine