Arksine/moonraker #967: missing data from files

MBogballe, 2025-02-21:
### What happened

after I updated one of my printers to Version 0.9.3-59-g62051108, it can't figure out how to read file data.
works fine in the older version 0.9.3-48 that I have on my other printers (I intentionally didn't update because it would come)

[image]

### Client

Mainsail

### Browser

Chrome

### How to reproduce

just updating to 0.9.3-59

### Additional information

[config-20250221-204053.zip](https://github.com/user-attachments/files/18914572/config-20250221-204053.zip)

Arksine, 2025-02-21:
Your config shows 3rd party modifications (`mmu_server`) are enabled.  I suspect the result is a duplicate of #964.  That said, all bug reports should be accompanied by `moonraker.log`, not `moonraker.conf`.

As with the noted issue, it will need to be corrected by the Happy Hare  developer.

MBogballe, 2025-02-21:
[moonraker (4).log](https://github.com/user-attachments/files/18914765/moonraker.4.log)

Arksine, 2025-02-21:
Confirmed.   The `mmu_server` module is enabled and it has monkey patched Moonraker to replace the file executed when metadata is extracted.  Moonraker should run `metadata.py`, as you can see your system is running `mmu_server.py`.  

```
2025-02-21 20:35:06,037 [shell_command.py:_check_proc_success()] - Command (/home/mb/moonraker-env/bin/python /home/mb/moonraker/moonraker/components/mmu_server.py -m -n -c /tmp/metacfg-_51pjt09.json) exited with return code 2
2025-02-21 20:35:06,038 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/mb/moonraker/moonraker/components/file_manager/file_manager.py", line 2566, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/mb/moonraker/moonraker/components/file_manager/file_manager.py", line 2621, in _run_extract_metadata
    raise self.server.error("Extract Metadata returned with error")
moonraker.utils.exceptions.ServerError: Extract Metadata returned with error
```

If you disable `[mmu_server]`, restart the moonraker service, and re-upload the file metadata will be correctly extracted.

MBogballe, 2025-02-21:
ok
so I'm surprised they haven't changed it since there was a new update from Happy yesterday

Arksine, 2025-02-21:
That will work.  You will probably need to ssh in to restart the `moonraker` service.  The UI may just do  soft restart which wont clear the changes made by monkey patching the code.

MBogballe, 2025-02-21:
great then it works again
it just had to be turned off completely to understand the change

thanks a lot for the help

Arksine, 2025-02-22:
YW.  I would expect HH to resolve the issue on their end in the near future.

closed 2025-02-22 by Arksine