Arksine/moonraker #971: Requesting history with date range during a print results in error

crysxd, 2025-03-01:
### What happened

Requesting the print history with any date range results in an error `'<' not supported between instances of 'NoneType' and 'float'`. It seems like the end date of the active print is `None` and is then compared with the date range provided in the request. I think the correct behavior should be to default the `None` to "now", which would include the active print if the requested date range spans "now"

### Client

Other

### Browser

Other or N/A

### How to reproduce

- Start a print
- Request `GET /server/history/list?limit=50&start=50&since=0&before=1898580550&order=asc`

Actual: `'<' not supported between instances of 'NoneType' and 'float'`
Expected: Results returned (including active print?)

### Additional information

I'm the OctoApp dev, this surfaced in the app. Would be cool if you could add OctoApp to the clients list in this bug report template :)

Arksine, 2025-03-01:
Thanks.  Can you attach `moonraker.log` reproducing the issue?

crysxd, 2025-03-01:
There is nothing in the logs besides the error message I mentioned

[moonraker.log](https://github.com/user-attachments/files/19038940/moonraker.log)

Arksine, 2025-03-01:
FWIW, I added OctoApp to the client list per your request.

crysxd, 2025-03-01:
Hm, I see there is an update for Moonraker. I did do an update session about 2 weeks ago, so I'm a bit surprised it's that old. 
After updating to `v0.9.3-64-g4889a2dd` the issue is gone...

I do agree with your opinion, it's better to not be included.

Sorry for the false alarm! :D
And thank you for adding OctoApp :)

closed 2025-03-01 by crysxd