Arksine/moonraker #977: Add Kalico support

CarVac, 2025-03-07:
### Is your feature request related to a problem? Please describe

When using Kalico, Moonraker complains that the repository (`https://github.com/KalicoCrew/kalico`) and branch names (`main` and `bleeding-edge-v2`) are invalid.

### Describe the solution you'd like

Recognize when the user is using Kalico's repositories and those two branches, and don't complain about them being invalid.

rogerlz, 2025-04-26:
If @Arksine is open to supporting it, I don't mind working on a PR for this.

Arksine, 2025-04-27:
I'm not entirely sure what is being requested here.  I presume its referring to the update manager?  In that case, forks and alternate branches are not tagged by Moonraker as "invalid".  Repos on unofficial remote/branches are considered as anomalies by Moonraker, as Moonraker only guarantees support for the official repo.  That said, updates are allowed to proceed.  If its just the messages that you would like to silence I suspect it could be done by the front end.

I have no intention of making life difficult for forks, however I would like to avoid adding any specific code for them.  If I do so for one I think it would be expected that I make accommodations for others as well.

AlejandroRivera, 2025-04-28:
Back when i reported this issue to the KalicoCrew in March 7th, 2025, what i saw in the update manager was the following:

[image]

which produced this other error when trying to do a soft recovery:

[image]

Having said that, i just checked again and this is what i see currently:

[image]

and when i click on the "Update" button, i see this, which seems correct:

[image]

I guess this got fixed somewhere in between then and now.

rogerlz, 2025-04-28:
> I'm not entirely sure what is being requested here. I presume its referring to the update manager

Yes, my apologies, we weren't clear enough.

> I have no intention of making life difficult for forks, however I would like to avoid adding any specific code for them.

I totally understand that.

> If its just the messages you want silenced I'd be open to a generic option that allows for that.

I like that, and my generic suggestion is to expose more to `[update_manager klipper]`, e.g., `git_repo` and `primary_branch`.
Further interactions of this "feature" would also be to expose `virtualenv`, `path`, `managed_services`. It would allow forks to move from `~/klipper` to an alternative location and systemd service.

cc @meteyou @pedrolamas

Arksine, 2025-04-29:
> I like that, and my generic suggestion is to expose more to [update_manager klipper], e.g., git_repo and primary_branch.

I could expose those options as an alternative to adding a new option.  Since these values are exclusively used for detecting anomalies, it no longer becomes necessary to provide a `report_anomalies` option if they can be user defined.  Of course there are other anomalies (such as untracked files), however upon further thought these can probably be hidden through front end customization.

> Further interactions of this "feature" would also be to expose virtualenv, path, managed_services. It would allow forks to move from ~/klipper to an alternative location and systemd service.

That shouldn't be necessary.  The values in `update_manager/common.py` are initial defaults.  Each time Moonraker connects to Klipper it will query the `info` endpoint for the current path, vitualenv, and process ID.  The paths are updated in the database, and the PID is used to determine the systemd unit name.  Feel free to experiment with it and provide feedback, however in  my testing it functions as expected.

Arksine, 2025-09-05:
The `report_anomalies` option is set per repo.  So it can be set to `false` for Klipper and `true` (the default) for other repos.  It can be set to `false` for any repo you wish to ignore anomalies.