Arksine/moonraker #978: import _ssl results in UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe7 in position 0: invalid continuation byte

dbarwikowski, 2025-03-08:
### What happened

After update to latest version of moonraker using kiauh moonraker doesn't start.
I was behind ~10 commits.

### Client

Mainsail

### Browser

Brave

### How to reproduce

I'm using official MainsailOS from Raspberry pi imager.
I've instalelled it on RPi Zero W

dbarwikowski, 2025-03-08:
It keeps restarting and no recent logs 

```
Is Dirty: False
Is Detached: False
Is Shallow: False
Commits Behind Count: 0
Diverged: False
Pinned Commit: None
2025-03-08 11:27:09,178 [base_deploy.py:log_info()] - Git Repo moonraker: Channel: dev
2025-03-08 11:27:09,179 [base_deploy.py:log_info()] - Git Repo moonraker: Validity check for git repo passed
2025-03-08 11:27:09,850 [server.py:_handle_term_signal()] - Exiting with signal SIGTERM
2025-03-08 11:27:10,006 [websockets.py:on_close()] - Websocket Closed: ID: 2945751480 Close Code: 1001, Close Reason: Server Shutdown, Pong Time Elapsed: 9.33
2025-03-08 11:27:10,022 [klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed
2025-03-08 11:27:10,378 [server.py:main()] - Server Shutdown
```

```
● moonraker.service - API Server for Klipper SV1
     Loaded: loaded (/etc/systemd/system/moonraker.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sat 2025-03-08 13:32:33 CET; 8s ago
       Docs: https://moonraker.readthedocs.io/
    Process: 796 ExecStart=/home/AnetA8/moonraker-env/bin/python $MOONRAKER_ARGS (code=exited, status=1/FAILURE)
   Main PID: 796 (code=exited, status=1/FAILURE)
        CPU: 1.115s
```

dbarwikowski, 2025-03-08:
It also broke kiauh v6
It throw error about UTF-8 and ssl

```
File "/usr/lib/python3.9/ssl.py", line 98, in <module>%0D%0A import _ssl # if we can't import it, let the error propagate%0D%0AUnicodeDecodeError: 'utf-8' codec can't decode byte 0xe7 in position 0: invalid continuation byte 
```

Arksine, 2025-03-08:
The output from the following command will provide more info:

```
sudo journalctl -xe -u moonraker.service
```

I suspect something went wrong with the git pull and the repo got corrupted.

> It also broke kiauh v6

Moonraker likely has nothing to do with this.  Its part of the Python standard library and Moonraker updates do not change it.  However it may be the reason Moonraker isn't starting as well.  It indicates that your filesystem may be corrupted.

dbarwikowski, 2025-03-08:
Hey @Arksine, thank you for your reply.
Everything was working fine, then I updated moonraker from Mainsail and it didn't reconnect after update.
Here is the log:
```
Mar 08 17:22:01 mainsailos systemd[1]: Started API Server for Klipper SV1.
░░ Subject: A start job for unit moonraker.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit moonraker.service has finished successfully.
░░
░░ The job identifier is 430.
Mar 08 17:22:02 mainsailos systemd[1]: moonraker.service: Main process exited, code=killed, status=11/SEGV
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit moonraker.service has exited.
░░
░░ The process' exit code is 'killed' and its exit status is 11.
Mar 08 17:22:02 mainsailos systemd[1]: moonraker.service: Failed with result 'signal'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit moonraker.service has entered the 'failed' state with result 'signal'.
Mar 08 17:22:12 mainsailos systemd[1]: moonraker.service: Scheduled restart job, restart counter is at 5.
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ Automatic restarting of the unit moonraker.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Mar 08 17:22:12 mainsailos systemd[1]: Stopped API Server for Klipper SV1.
░░ Subject: A stop job for unit moonraker.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A stop job for unit moonraker.service has finished.
░░
░░ The job identifier is 636 and the job result is done.
Mar 08 17:22:12 mainsailos systemd[1]: Started API Server for Klipper SV1.
░░ Subject: A start job for unit moonraker.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit moonraker.service has finished successfully.
░░
░░ The job identifier is 636.
Mar 08 17:22:14 mainsailos systemd[1]: moonraker.service: Main process exited, code=killed, status=11/SEGV
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit moonraker.service has exited.
░░
░░ The process' exit code is 'killed' and its exit status is 11.
Mar 08 17:22:14 mainsailos systemd[1]: moonraker.service: Failed with result 'signal'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit moonraker.service has entered the 'failed' state with result 'signal'.
Mar 08 17:22:14 mainsailos systemd[1]: moonraker.service: Consumed 1.002s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit moonraker.service completed and consumed the indicated resources.
```

dbarwikowski, 2025-03-08:
I've used kiauh v5 to remove moonraker. Now I have segmentation fault when installing:
```
###### Installing Moonraker ...

###### Checking for the following dependencies:
● git
● wget
● curl
● unzip
● dfu-util
● virtualenv
● libjpeg-dev
● zlib1g-dev
[✓ OK] Dependencies already met!

###### Cloning Moonraker from https://github.com/Arksine/moonraker.git ...
Cloning into '/home/AnetA8/moonraker'...
remote: Enumerating objects: 11631, done.
remote: Counting objects: 100% (15/15), done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 11631 (delta 4), reused 3 (delta 3), pack-reused 11616 (from 2)
Receiving objects: 100% (11631/11631), 4.42 MiB | 580.00 KiB/s, done.
Resolving deltas: 100% (8675/8675), done.

###### Installing dependencies ...

###### Reading dependencies...
python3-virtualenv
python3-dev
libopenjp2-7
libsodium-dev
zlib1g-dev
libjpeg-dev
packagekit
wireless-tools
curl
build-essential

###### Installing packages...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9).
curl is already the newest version (7.74.0-1.3+deb11u14).
libjpeg-dev is already the newest version (1:2.0.6-4).
libopenjp2-7 is already the newest version (2.4.0-3).
libsodium-dev is already the newest version (1.0.18-1).
packagekit is already the newest version (1.2.2-2).
python3-dev is already the newest version (3.9.2-3).
python3-virtualenv is already the newest version (20.4.0+ds-2+deb11u1).
wireless-tools is already the newest version (30~pre9-13.1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-2+deb11u2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[✓ OK] Moonraker packages installed!

###### Installing python virtual environment...
/home/AnetA8/kiauh/scripts/moonraker.sh: line 332:  1692 Segmentation fault      virtualenv -p /usr/bin/python3 "${MOONRAKER_ENV}"
>>>>>> Creation of Moonraker virtualenv failed!
```

Arksine, 2025-03-08:
Moonraker isn't even getting a chance to start per the journal, it is seg faulting as well.  The type of failures you are getting (import error on Python library, seg fault running virtualenv and Moonraker) are indicative of file system corruption.   Your SD Card may be failing.  The other possibility is the CPU on the Pi isn't stable.

You can try to reimage, if you have a new SD Card I would advise using one.  It may be possible for you to save your `~/printer_data` folder since you can still SSH into the device.  Just use scp to copy it to your local PC.

dbarwikowski, 2025-03-08:
Well, I didn't expect that outcome but you might be right. I get more and more weird errors

Arksine, 2025-03-08:
It can happen.  When a file system starts failing any action that writes to the disk can lead to more failures and instability.  Just happened to occur after an update.

It sounds like its getting worse.  I would try to copy the `printer_data` folder from that Pi as soon as possible, presuming you want to restore it.