Arksine/moonraker #989: Klipper Estimator config dump fails with reverse proxy

bluecube, 2025-04-22:
### What happened

Dumping  config for Klipper Estimator fails for me, because internally it tries to access `http://192.168.1.31:7125` while my moonraker instance is listening only on `localhost:7125` (reverse-proxied to port 80 afterwards). If the HTTP request was made to `localhost:7125`, or `192.168.1.31:80`,  this would have worked for me.

I have tried to modify the moonraker conf to listen on the default `0.0.0.0`, then klipper_estimator works.  I tried to look into `analysis.py`, but I couldn't find a straightforward fix with my (very shallow) understanding of moonraker :-).

I understand that this configuration might not be entirelly supported by the project, but at the same time it feels like a bit of an unnecessary limitation, when it is basically moonraker accessing its own API (If I understand it correctly :) ).

### Client

Fluidd

### Browser

Firefox

### How to reproduce

- Set `host: localhost` and enable `[analysis]` section in config
- Setup reverse proxy to access the API for external control (or only test locally?)
- Klipper Estimator config dump does not work

### Additional information

[moonraker.log](https://github.com/user-attachments/files/19845349/moonraker.log)

Arksine, 2025-04-22:
Thanks.  You are correct, this is a bug.  The public IP shouldn't be used when Moonraker is configured to listen on the localhost.  Commit 1c124149720e24f52b80ce82ce906e69476b7724 should resolve this.

bluecube, 2025-04-22:
Thanks for the quick fix, it work now