Arksine/moonraker #995: Register /tmp as a root?

voney, 2025-06-12:
### Is your feature request related to a problem? Please describe

Hi, I'm in the early stages of writing a plugin and I'm trying to figure out how to read the printer's `/tmp` directory via the moonraker API.

The documentation talks about directory roots being "registered" but gives not indication as to how this could be done. Only allowing one to see which roots have been registered.

### Describe the solution you'd like

Ideally I'd like a configuration option in the `moonraker.conf` that allows one to configure one or more additional roots.

```
[file_manager_root myfiles]
path: ~/myfiles
enable_write_access: True

[file_manager_root temp]
path: /tmp
enable_write_access: False
```

### Describe alternatives you've considered

- Try to get other extensions and plugins to write to an already exposed directory like `logs`
  - Means having to do multiple PRs accross multiple repos.
  - Default klipper functions write to `/tmp` seems like not exposing it may be an anti-pattern.

Laikulo, 2025-07-29:
You probably want something like config.get_server().lookup_component("file_manager").register_directory(...)`
https://github.com/Arksine/moonraker/blob/77c91493488ea04af35ae94d6158a4e83ab75619/moonraker/components/file_manager/file_manager.py#L272

It takes 'root', the API name the dir will be accessed by, then the path to expose, and finally a bool for enabling full/write access.
for example `register_directory('tmp','/tmp',False)

voney, 2025-07-29:
@Laikulo Thanks for the feedback, that's useful if I'm trying to write a plugin within moonraker. I'm actually trying to write a plugin for fluidd and just need to be able to access other directories. Hence this being a feature request since I think this base functionality of being able to have the pre-existing moonraker file API work with other directories via config rather than needing a separate moonraker plugin to allow this.

To be clear, I am 100% down to write this myself, but I don't want to bother if it's something the moonraker team wouldn't consider up-streaming.

voney, 2025-07-30:
@Arksine Thanks for the suggestion. Unfortunately I don't think asking users of my plugin to create symlinks is a solution. 

The core thing I'm trying to do here is to create a fluidd plugin that shows input shaper information loaded from CSV files in `/tmp` as far as I'm aware the `/tmp` location is [hard-coded in klipper](https://www.klipper3d.org/G-Codes.html?h=test_resonances#test_resonances)

In my humble opinion, `/tmp` should already be available via moonraker's API since klipper uses this directory for input shaper output, and the fact it's not is actually a bug. BUT my suggestion here is to make it a configurable thing via `moonraker.conf` to appease those who disagree.

Like I said, I'm happy to code this up, but I want some clarity on whether the moonraker team would even accept it. As it stands I don't believe the moonraker API is feature complete vs klipper functionality.

Arksine, 2025-07-30:
> In my humble opinion, /tmp should already be available via moonraker's API since klipper uses this directory for input shaper output, and the fact it's not is actually a bug. BUT my suggestion here is to make it a configurable thing via moonraker.conf to appease those who disagree.

Respectfully I disagree that it is bug.  A bug is a defect, ie: existing functionality is behaving in a way that is not intended.   Missing functionality would only fall into this category if it was intended to be present.   I find the argument that it would be useful to register `/tmp` to be reasonable, although I think it would be preferred if Klipper didn't hardcode the `/tmp` location.  Some systems do not register `/tmp` as the temporary file location.

> Like I said, I'm happy to code this up, but I want some clarity on whether the moonraker team would even accept it. As it stands I don't believe the moonraker API is feature complete vs klipper functionality.

Moonraker will never be "feature complete" in this sense.  There will always be features that one or more users would like to have that will not be available.  When adding new features there several considerations must be weighed...exposure, performance impact, broad appeal, complexity, and maintenance cost come to mind.

As mentioned in my previous post there are several good reasons for restrictions on root registration, therefore configurable roots are not something I would consider merging.     That said, as a compromise I am willing to register the `/tmp` folder as a read-only root if it exists and the Moonraker user has read permission.  There shouldn't be any sensitive files in `/tmp` so exposure should be minimal and registering as read-only mitigates any potential performance issues introduced by inotify.

voney, 2025-08-01:
@Arksine thanks for following up. Apologies if I came across as an ass, I appreciate you taking this request seriously.