Arksine/moonraker #998: tplink_smartplug fails to power on KP303 plugs

lrosselld, 2025-06-26:
### What happened

My [power section] in moonraker config was working until it wasn't.

I have the correct ip address and port # and I can confirm as this was the same config as I had when it was working. 

### Client

Fluidd

### Browser

Brave

### How to reproduce

Include the following in the moonraker.conf:

[power pi]
type: tplink_smartplug
address: (static ip of powerstrip)
output_id: 2
off_when_shutdown: True

[power printer]
type: tplink_smartplug
address: (static ip of powerstrip)
output_id: 1
initial_state: on
restart_klipper_when_powered: True

When powering on the [power pi] device, I get the following error in moonraker.log:

2025-06-25 19:37:19,963 [server.py:_initialize_component()] - Performing Component Post Init: [power]
2025-06-25 19:37:19,963 [server.py:_initialize_component()] - Performing Component Post Init: [update_manager]
2025-06-25 19:37:20,220 [power.py:init_state()] - Power Device printer: setting initial state to on
2025-06-25 19:37:20,300 [power.py:set_power()] - Power Toggle Error: printer
Traceback (most recent call last):
  File "/home/dude/moonraker/moonraker/components/power.py", line 962, in set_power
    res = await self._send_tplink_command(state)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dude/moonraker/moonraker/components/power.py", line 842, in _send_tplink_command
    sysinfo = await self._send_tplink_command("info")
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: coroutine ignored GeneratorExit
2025-06-25 19:37:20,306 [base_events.py:default_exception_handler()] - Task was destroyed but it is pending!
task: <Task pending name='Task-23' coro=<TPLinkSmartPlug.init_state() done, defined at /home/dude/moonraker/moonraker/components/power.py:912> wait_for=<Future pending cb=[Task.task_wakeup()]>>

[moonraker.log](https://github.com/user-attachments/files/20915909/moonraker.log)

Arksine, 2025-06-27:
This should be fixed with the latest commits.

lrosselld, 2025-06-29:
> This should be fixed with the latest commits.

Can confirm this issue is now resolved.

Thank you @Arksine your work is highly appreciated! I sent you something for coffee!

closed 2025-06-29 by lrosselld