Arksine/moonraker #1002: moonraker not updating 0.9.3.75 -> 0.9.3.85

wolfcharm, 2025-07-06:
### What happened

error occur during update process: 
`name: expected 'pillow', but metadata has 'unknown'`

then downloading goes:
`Downloading Pillow-10.1.0.tar.gz`
then intallation and again

```
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'done'
Requirement already satisfied: pyserial==3.4 in /usr/lib/python3.8/site-packages (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 4)) (3.4)
Collecting pillow<=11.1.0,>=9.5.0 (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 5))
Using cached pillow-10.4.0.tar.gz (46.6 MB)
Installing build dependencies: started
```
```
moonraker/components/sensor.py                    |  3 +-
moonraker/components/shell_command.py             |  9 ++--
moonraker/components/spoolman.py                  |  2 +-
moonraker/components/template.py                  |  5 +-
moonraker/components/update_manager/app_deploy.py |  1 +
pyproject.toml                                    |  5 +-
scripts/moonraker-requirements.txt                |  5 +-
18 files changed, 153 insertions(+), 67 deletions(-)
Git Repo moonraker: Checking pip version...
Git Repo moonraker: Pip version up to date
Git Repo moonraker: Updating python packages...
Looking in links: /usr/data/moonraker/moonraker/scripts/python_wheels
Looking in links: /usr/data/moonraker/moonraker/scripts/python_wheels
Ignoring importlib_metadata: markers 'python_version == "3.7"' don't match your environment
Requirement already satisfied: tornado<=6.5.1,>=6.2.0 in /usr/data/moonraker/moonraker-env/lib/python3.8/site-packages (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 3)) (6.4.1)
Requirement already satisfied: tornado<=6.5.1,>=6.2.0 in /usr/data/moonraker/moonraker-env/lib/python3.8/site-packages (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 3)) (6.4.1)
Collecting tornado<=6.5.1,>=6.2.0 (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 3))
Collecting tornado<=6.5.1,>=6.2.0 (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 3))
Downloading tornado-6.4.2.tar.gz (501 kB)
Downloading tornado-6.4.2.tar.gz (501 kB)
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Preparing metadata (pyproject.toml): finished with status 'done'
Requirement already satisfied: pyserial==3.4 in /usr/lib/python3.8/site-packages (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 4)) (3.4)
Requirement already satisfied: pyserial==3.4 in /usr/lib/python3.8/site-packages (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 4)) (3.4)
Collecting pillow<=11.1.0,>=9.5.0 (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 5))
Collecting pillow<=11.1.0,>=9.5.0 (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 5))
Downloading pillow-10.4.0.tar.gz (46.6 MB)
Downloading pillow-10.4.0.tar.gz (46.6 MB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 46.6/46.6 MB 1.4 MB/s eta 0:00:00
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 46.6/46.6 MB 1.4 MB/s eta 0:00:00
Installing build dependencies: started
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 46.6/46.6 MB 1.4 MB/s eta 0:00:00
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 46.6/46.6 MB 1.4 MB/s eta 0:00:00
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Installing backend dependencies: started
Installing backend dependencies: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'done'
Discarding https://files.pythonhosted.org/packages/ef/43/c50c17c5f7d438e836c169e343695534c38c77f60e7c90389bd77981bc21/pillow-10.3.0.tar.gz (from https://pypi.org/simple/pillow/) (requires-python:>=3.8): Requested unknown from https://files.pythonhosted.org/packages/ef/43/c50c17c5f7d438e836c169e343695534c38c77f60e7c90389bd77981bc21/pillow-10.3.0.tar.gz (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 5)) has inconsistent name: expected 'pillow', but metadata has 'unknown'
Discarding https://files.pythonhosted.org/packages/f8/3e/32cbd0129a28686621434cbf17bb64bf1458bfb838f1f668262fefce145c/pillow-10.2.0.tar.gz (from https://pypi.org/simple/pillow/) (requires-python:>=3.8): Requested unknown from https://files.pythonhosted.org/packages/f8/3e/32cbd0129a28686621434cbf17bb64bf1458bfb838f1f668262fefce145c/pillow-10.2.0.tar.gz (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 5)) has inconsistent name: expected 'pillow', but metadata has 'unknown'
Downloading Pillow-10.1.0.tar.gz (50.8 MB)
Looking in links: /usr/data/moonraker/moonraker/scripts/python_wheels
Ignoring importlib_metadata: markers 'python_version == "3.7"' don't match your environment
Requirement already satisfied: tornado<=6.5.1,>=6.2.0 in /usr/data/moonraker/moonraker-env/lib/python3.8/site-packages (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 3)) (6.4.1)
Collecting tornado<=6.5.1,>=6.2.0 (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 3))
Using cached tornado-6.4.2.tar.gz (501 kB)
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'done'
Requirement already satisfied: pyserial==3.4 in /usr/lib/python3.8/site-packages (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 4)) (3.4)
Collecting pillow<=11.1.0,>=9.5.0 (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 5))
Using cached pillow-10.4.0.tar.gz (46.6 MB)
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Installing backend dependencies: started
Installing backend dependencies: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'done'
Discarding https://files.pythonhosted.org/packages/cd/74/ad3d526f3bf7b6d3f408b73fde271ec69dfac8b81341a318ce825f2b3812/pillow-10.4.0.tar.gz (from https://pypi.org/simple/pillow/) (requires-python:>=3.8): Requested unknown from https://files.pythonhosted.org/packages/cd/74/ad3d526f3bf7b6d3f408b73fde271ec69dfac8b81341a318ce825f2b3812/pillow-10.4.0.tar.gz (from -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt (line 5)) has inconsistent name: expected 'pillow', but metadata has 'unknown'
Using cached pillow-10.3.0.tar.gz (46.6 MB)
Installing build dependencies: started
```

### Client

Fluidd

### Browser

Microsoft Edge

### How to reproduce

update from 0.9.3.75 to 0.9.3.85

Arksine, 2025-09-14:
Please attach a full `moonraker.log`.  At a glance this appears to be a system issue with pip.  Judging by the snippet I'm guessing this is Creality hardware, which IMO was a poor choice on their behalf and not sufficient to run Moonraker.  Python 3.8 has been EOL for a year and it seems they provide no easy path to upgrade.

The only thing I can potentially do in Moonraker is bump the pinned version of pip.  I'm not sure this will resolve the issue though.  You may be able to upgrade pip manually by running the following command:

```
/usr/data/moonraker/moonraker-env/bin/pip install -U pip
```

The-Monkey-King, 2025-09-16:
I know it is a pain in the ass, but resetting to factory and then re-installation does work.

And yes, my issue came from Creality K1 Max. 

I think the main issue probably is, as Sineos said in the Klipper Discourse Group, that companies are taking Klipper and modding it to fit their needs without publishing their fork and fairly attributing Klipper. Guilouz's HelperScript allows for some Klipper addons but it follows Creality's format and not Klipper's. 

I'd love to have another machine so that I could "greenfield" the whole OS and Klipper environments but unfortunately I rely on my sole K1 Max for a ton of production work. So all I can do is keeping it working even though we may sooner or later, have to deal with maintaining downlevel software/operating system.