Arksine/moonraker #1009: Issue saving history

mjfsch, 2025-08-01:
### What happened

Not too sure what this means, but i no longer get history saving correctly.
The job marks as completed on mainsail and klipper screen, but never appears in the history list via mainsail, and the gcode file never shows the green tick next to it to show it has been printed previously.

seems related to adding a start print macro that calls a pause at the start of print to home filament needed on a multi filament setup
maybe caused by CLEAR_PAUSE been used?

### Client

Mainsail

### Browser

Chrome

### How to reproduce

print, then check history.

### Additional information

suspect this is the error

[logs-20250801-151213.zip](https://github.com/user-attachments/files/21549794/logs-20250801-151213.zip)

```
2025-08-01 13:54:57,985 [file_manager.py:get_file_list()] - Updating File List <gcodes>...
2025-08-01 13:55:11,552 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 127.0.0.1
2025-08-01 13:55:11,553 [web.py:log_exception()] - Uncaught exception GET /server/history/job?uid=None (127.0.0.1)
HTTPServerRequest(protocol='http', host='127.0.0.1:7125', method='GET', uri='/server/history/job?uid=None', version='HTTP/1.1', remote_ip='127.0.0.1')
Traceback (most recent call last):
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/tornado/web.py", line 1848, in _execute
    result = await result
  File "/home/biqu/moonraker/moonraker/components/application.py", line 683, in get
    await self._process_http_request(RequestType.GET)
  File "/home/biqu/moonraker/moonraker/components/application.py", line 700, in _process_http_request
    result = await self.api_defintion.request(
  File "/home/biqu/moonraker/moonraker/components/history.py", line 275, in _handle_job_request
    f"SELECT * FROM {HIST_TABLE} WHERE job_id = ?", (int(job_id, 16),)
ValueError: invalid literal for int() with base 16: 'None'
2025-08-01 13:55:11,557 [application.py:log_request()] - 500 GET /server/history/job?uid=None (127.0.0.1) [_TRUSTED_USER_] 6.49ms
```

mjfsch, 2025-08-02:
Thanks for looking at this.
Testing the delay idea now.

Arksine, 2025-08-02:
Update:  I just pushed commit f20feff6c67884ff1253414506c43222e04c0460, which should work around this issue even if no delay is added before the pause.

closed 2025-08-02 by mjfsch