Arksine/moonraker #1010: [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found

BeeKeeper74, 2025-08-12:
### What happened

Hi

I updated all Klipper components of 3D printer via the update all button.

Components updated to versions:
fluidd v1.34.3
klipper  v0.13.0-192-g3ef760c1
KlipperScreen  v0.4.6-7-gc8a3a15c
moonraker v0.9.3-102-gf20feff6

After updating I see the error "Klippy not connected."

moonraker.log contains errors:
2025-08-12 16:24:46,499 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found
2025-08-12 16:24:48,500 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found
2025-08-12 16:24:50,498 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found
2025-08-12 16:24:52,498 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found
2025-08-12 16:24:54,587 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found

klippy.log have no errors.

Please help. 

[moonraker.log](https://github.com/user-attachments/files/21736795/moonraker.log)

### Client

KlipperScreen, Fluidd

### Browser

Chrome

### How to reproduce

Update to
fluidd v1.34.3
klipper  v0.13.0-192-g3ef760c1
KlipperScreen  v0.4.6-7-gc8a3a15c
moonraker v0.9.3-102-gf20feff6

disdkein, 2025-08-20:
I ran into a similar problem myself yesterday. what preceded the error?

HXGdiaz, 2025-08-20:
Updated and got the same problem Klippy not connected.

closed 2025-08-25 by BeeKeeper74